ThunderFly-aerospace/TF-G2 #2: Nosná podložka

roman-dvorak, 2019-08-31:
Nosná podložka malého vírníku se bude skládat ze dvou identických částí. 

Budou to 2 destičky, které se následně vzájemně sešroubují pomocí nějaké vytištěné spojky. 
Destička bude mít podélně vyšší límec, který bude mít dva úkoly. První je zpevnění celé konstrukce. Druhý je vytvoření míst pro připevnění bočních stěn. 

Podložka bude 55 mm široká a dlouhá 2x140 mm. Tato délka byla zvolena z technologických důvodů - maximální tiskový rozměr.

[image]

Příčný řez dílem
[image]

Ve spodní části budou díry na šrouby na připevnění dalšího příslušenství. Budou zarovnané do nějaké určité mříže. Předpokládám nějakých 4-5 řad. Budou to díly na M3 šrouby.

Všechny rozměry musí být udělány pomocí proměnných parametricky.

roman-dvorak, 2020-04-02:
Nosná podložka je aktuálně ve stavu, že obsahuje boční lem s řadou děr pro připevnění bočnic. Podložka bude dělena na 2 časti. Na obrázku to je naznačeno. Bočnice budou na 3 časti a je potřeba zajistit, aby předěl částí nevycházel na stejné místo. 

Od původního modelu došlo k odlehčení vytvořením vetších otvorů ve dnu modelu. Tyto otvory bude potřeba upravit v návaznosti na připevnění podvozku. Může pak být provedeno další odlehčení. Nebo ztečení či chytřejší vyztužení. 

Na začátek první zprávy jsem vypsat věci k úpravě. 

[image]

roman-dvorak, 2020-04-25:
Do podložky přidány šestihrany na zapuštění matky, upraveno odlehčení a otvory pro připevňení ocasu a předního podvozku.

kaklik, 2020-05-18:
Ten způsob připevnění předního podvozku na dva šrouby je podle mě vyhovující.

closed 2020-05-25 by roman-dvorak