ThunderFly-aerospace/TF-G2 #7: 888_1007 a 888_1008, pylon

roman-dvorak, 2020-04-02:
-[ ] Základní model
---
Pylon byl namodelován po vzoru modelu virníku Auto-G2. Jedná se tedy o 2 placky z boků vírníku sbíhající se do středu k rotorové hlavě. Každá část bude uprostřed sešroubována ke středovému rámečku. Vice v samostatném issue. 

Stěny pilonu budou umístěny šikmo. Proto je vhodné aby na koncích byly klíny. Na spodní časti je to už implementováno.

[image]

roman-dvorak, 2020-04-10:
Aktuální návrh pilou vypadá takto:

[image]

[image]

[image]

Pro první létací prototyp mi to přijde dostatečné.

kaklik, 2020-04-11:
Má to nějakou použitelnou tuhost?

roman-dvorak, 2020-04-11:
Horní část je dostatečně tuhá. U spodní jsem přemýšlel o nějaké příčce popřípadě přímodelování výztuhy. Ale to bych řešil později.

Spodní část nemůžu moc otestovat, protože nemám vytištěnou podložku ve správném rozměru. 

Jinak jsem nepsal. Celé (tak, jak je to na fotce) má 77g

Ještě jsem přemýšlel o posunutí osy rotoru trochu dopředu. Aby to bylo v ose pylonu a nevznikal tam takový moment, který by tomu pomáhal se kroutit. Ale to také až do další verze.

roman-dvorak, 2020-04-11:
Jak je ta střední příčka, tak tam jsou připevňovací otvory pro Venturiho trubici.

closed 2020-04-25 by roman-dvorak