ThunderFly-aerospace/TF-G2 #9: Zadní podvozek - tuhost odpružení

kaklik, 2020-04-06:
Aktuální návrh zaního dvoukolého podvozku vypadá takto.

[image]

Jeho hlavním problémem je nízká tuhost. Zároveň i současná poměrně vysoká hmotnost podvozku není úplně ideální.  Proto je potřeba tuhost zvýšit při zachování současného množství použitého materiálu. 

Řešením tohoto problému může být vytvoření "prolisů" v pružinách podvozku.

roman-dvorak, 2020-04-26:
Na vírníku TF-G1 bude tento díl použit jako přední podvozek.

kaklik, 2020-05-05:
[image]

Návrh nového tvaru je správný.  Jen jím je teď potřeba nahradit původní rovný materiál, alespoň v nějaké části označené úhlem alfa na nákresu výše. 
Pro zjednodušení je ale asi možné tímto tvarem střechy nahradit celou tu část čtvrtkruhu - nejhorší co se stane, je že  ten podvozek bude příliš tuhý.

kaklik, 2020-05-26:
Aktuálně v modelu vidím tohle: 

[image]

Což je přesně to, co je potřeba. :)

kaklik, 2020-05-28:
Tak konstrukční změna je znatelným zlepšením. 
[image]

Ta tuhost ale stále zřejmě není dostatečná, takže musíme vymyslet, jak to zlepšit. :)

roman-dvorak, 2020-05-29:
@RomanDCZ přišel s návrhem podvozku načrtnutém na následujícím obrázku

[image]

Stěny by možná mohly být ještě tenčí než jsou v aktuálním návrhu. Ale bylo by potřeba to zase otestovat.

kaklik, 2020-05-29:
Před návrhem tohoto bych možná ještě navrhoval zkusit více prolomit ten prolis v současném návrhu. Protože podle fotky mi to přijde stále dost placaté. 

Ty otvory v novém návrhu na obrázku výše by asi měly být zase trojúhelníkové podobně jako na trupu, aby se ty příčky nebortily.

RomanDCZ, 2020-05-29:
Čau Romy, ten náhled je trochu zavádějící, protože tam je výška toho
nosníku příliš velká. Když to tak odhadnu, tak u trupu bych měl výšku
nosníku cca 15mm, stěny s poloviční tloušťkou a výřezy, žebra mezi stěnami
asi tak, jak máš na obrázku. Když by to bylo moc tuhé, praskalo by to.

Přijdeš domů? Zavolej mamce. Měj se.

kaklik, 2020-06-20:
Protože se tohle issue moc nehýbe, tak by možná bylo lepší zvolit jinou technologii podvozku.  Další možnost je použití menších tištěných dílů a [pružinového drátu](https://obchod.lasermodely.cz/kovy/290-nerezovy-pruzinovy-drat.html#/97-prumer-3_mm).

V tomto případě by byly tištěny dva druhy dílů. Jeden pro uchycení podvozku k trupu vírníku a pak dva kusy dílu pro uchycení kolečka. Ty by v nejlepším případě měly být symetrické, aby nebylo nutné tisknout zvlášť díl pro levé a pravé kolečko.

Taková konstrukce by mohla být realizována následovně. Kdy dráty budou uříznuty na správnou délku a zalepí se do vytištěných dílů.  

[image]

Každé kolečko by bylo uchyceno na dvojici drátů, Tím by jednak mělo být možné použít tenčí drát a zároveň se tím zvýší předozadní tuhost podvozku, což je záležitost s kterou měl problém vírník [Auto-G2](https://github.com/ThunderFly-aerospace/Auto-G2).

Při návrhu dílů je však nutné dbát na správné vedení tiskových vrstev.  Díl kolečka musí být tištěn tak, aby tiskové vrstvy byly v ose drátu. Tato tisková pozice je v obrázku naznačena v pravo nahoře.
U horního dílu, který drží podvozek a který nelze moc tisknout na bok, tak aby vrstvy byly v osách podvozkových drátů,  by tak měly být oblé přechody mezi válcovými držáky podvozkových drátů a montážní plochou. Tím je alespoň rozloženo namáhání vrstev přes výšku několika vrstev.

kaklik, 2020-07-15:
Tak aktuálně navržená varianta, kterou kupodivu jde i docela snadno tisknout (až na jednu stranu, kde jsou prořezy hodně sesunuté). 

[image]

Tohle řešení má hodně zajímavé vlastnosti a v podstatě je kombinací přístupů navržených @Elizaa62  a @RomanDCZ . 

Tuhost tohoto řešení je podobná, nebo mírně větší, než prvního vytištěného návrhu. Neobsahuje ale proti původnímu návrhu téměř žádný materiál! 

Jsem si tak jistý, že nějakou změnou parametrů půjde dosáhnout ještě větší tuhosti.  Například by se mělo vyzkoušet: 

1. Ještě více prolomit střední část podvozku směrem na vnější stranu
2. Přidat symetrickou děrovanou strukturu i s vnitřní strany.

Elizaa62, 2020-08-31:
1. Prořezy jsou posouvané vždy o určitý úhel
2. prolomení si  můžeš nastavit sám - proměnná "angle" - teď jsem ji nastavila na 55°, ale mám problém v tom,že při zmenšení úhlu je nutné prodloužit i ty koncovky, protože ten materiál bude přesahovat přes koncovky - tak jsem doplnila proměnnou "presah"
3. Jak to myslíš?

kaklik, 2020-09-02:
Tak jsem vytiskl novou verzi: 

[image]

Je skutečně mnohem pevnější než předchozí, navíc jde vytisknout podstatně snadněji (zkoušel jsem tisknout dva kusy). 
Nové poznatky, které díky tomu máme jsou: 

1. Tuhost pro dosednutí vírníku je teď už myslím dostatečná
2. Zřejmě by bylo vhodné aby podvozková noha byla skutečně úplná čtvrtkružnice (nebo i trošku více)- Nyní je to tak, že chybí několik stupňů do stavu, že  kolečko bude bez zatížení rovně k povrchu. Teď je bez zatížení přikloněné k vírníku. Zde si dokonce myslím, že kolečka by bez zatížení měla být naopak mírně vykloněna, aby při zatížení podvozku čistě hmotností vírníku byla kolmo k zemi. 
3. Je možná problém s nízkou torzní tuhostí podvozku protože je poměrně snadné podvozkovou nohu takhle ohnout: 

[image]

To by ale možná mohla zlepšit myšlenka z bodu 3. výše. Kdy jsem zvažovat průřez podvozkovou nohou symetricky zdvojit. Nějak takhle:

Elizaa62, 2020-09-04:
2. pro délku té kružnice tam asi slouží ta funkce "suspension_camber" z hodnoty 10 jsem ji snížila na hodnotu 2.
--> přidala jsem i ty výřezy.

3. udělám po víkendu

kaklik, 2020-09-04:
Ad 2.  s tou délkou myslím budeme ještě několikrát hýbat.  Protože bude potřeba naladit délku tak, aby při zatížení vírníkem bylo kolečko právě kolmo k zemi.  Takže by se hodilo vyřešit nějak to automatický generování výřezů, aby se to nemuselo stále ručně kopírovat. ;)

Po víkendu budeme vědět více o tom jak se tohle řešení chová.  Dnes to vytisknu a někdy zítra namontujem na vírník.

kaklik, 2020-09-07:
Po montáži na vírník se ukázalo, že nová úprava podvozku má o hodně menší předozadní tuhost.  Podvozková noha se tak snadno kroutí ve směru otáčení koleček.  
Pomoct by tomu mohlo přidání symetrické vnitřní struktury zmíněné výše.   V případě, že to dostatečně nepomůže, tak by zřejmě šlo podvozkovou nohu nahoru směrem k trupu postupně rozšiřovat. 

[image]

[image]

Elizaa62, 2020-09-17:
Hotovo

kaklik, 2020-09-21:
Nové řešení s výztuhou z vnitřní strany funguje mnohem lépe a vírník na novém podvozku je bez problémů schopen stát. 

Potíž vzniká při přistání, kdy dojde ke zvýšenému předozadnímu namáhání podvozku. V takové situaci se obvykle podvozková noha vykroutí dozadu poblíž uchycení na trup vírníku. 

Řešením by zřejmě bylo postupné rozšíření podvozkové nohy od kolečka ke trupu, tak aby nejvíce na krut namáhaná část (u trupu), byla zároveň nejširší.  Rozšíření bych navrhoval realizovat jen v jedné ose ve směru délky trupu vírníku. 

[image]

slimonslimon, 2020-09-30:
[image]

je to vyrvane z puvodního zdrojaku a upravené - pozor i čast označená jako původní zdrojak je upravená.

[sesikmena_noha.txt](https://github.com/ThunderFly-aerospace/TF-G2/files/5304734/sesikmena_noha.txt)

Elizaa62, 2020-10-01:
Použila jsem Vítkův nápad.. bohužel mi funguje pouze pro jednu stranu - zelená část. Pro tu modrou část funguje, ale z mě nevysvětlitelného důvodu mi části nesedí k sobě.. a už fakt nevím, co s tím...

slimonslimon, 2020-10-02:
Zdá se mi, že to totiž není vůbec jednoduché, dokonce zda je to vůbec řešitelné s tou vnitřní stranou... ale možná to jen nevidím

rychlý a možná funkční workround by byl udělat tu vnitřní stranu palcatou (nevím zda to někdo zkoušel nebo ne, já mám něco takového na jedné verzi nohou udělané z tvrdého papíru) - že by řezem té nohy byl  trojúhelník... samozřejmně s prořezy, které papír nemá

[image]

slimonslimon, 2020-10-02:
aaha, tak, řešení existuje...

[sesikmena_noha_2.txt](https://github.com/ThunderFly-aerospace/TF-G2/files/5316715/sesikmena_noha_2.txt)

místo ořezávání excentrickým válcem, je potřeba ořezávat excentrickými kužely, které pak budou vnějším pláštěm toho vnitřku.

[image]

kaklik, 2020-10-02:
Aha super! :)

Ta vnitřní placatá strana, je sice workaround pro teď. Nicméně ta tuhost podvozku s placatou vnitřní stranou bude myslím nižší, než s tou příhradovou konstrukcí z vnitřní strany.  Neboť tady při kroucení podvozku se prakticky nemění ta příčná vzdálenost na vnitřní straně. 

Z toho mi vyplývá, že tu symetrickou vnitřní stranu potřebujem  (i kdyby měla mít tupější úhel než ta vnější).

slimonslimon, 2020-10-02:
zdá se, že úhel otočení, aby byla noha tisknutelná se dá zjistit z bočního nárysu, kdy kužele přejdou do trojůhelníků.

Docela mne děsí různé závislosti parametrů.

kaklik, 2020-10-02:
No ještě existuje jedna možnost jak tu podvozkovou nohu zkonstruovat.  Myslím, že by to šlo za použití téhle knihovny: https://www.thingiverse.com/thing:900137#files

Pak by průřez nohy nebyl kosočtverec, ale airfoil..    Problém s natočením nohy pro tisk se tím ale určitě neřeší, naopak zesložiťuje, neboť airfoil by pak musel být levý a pravý. Ještě k tomu by se zřejmě musela splacatit třeba náběžná nebo odtoková hrana, aby to bylo na co položit při tisku.

slimonslimon, 2020-10-02:
to je dle mého názoru cesta do pekla

kaklik, 2020-10-02:
> to je dle mého názoru cesta do pekla

Z jakého důvodu?

slimonslimon, 2020-10-04:
> > to je dle mého názoru cesta do pekla
> 
> Z jakého důvodu?

Po osbní konzultaci vyšlo najevo, že jsem úplně nepochopil jak chtěl @kaklik tu knihovnu použít.

kaklik, 2020-10-05:
@Elizaa62 @slimonslimon Pokračujte prosím nějak ve spolupráci na řešení toho podvozku tou původní metodou.   Já se výhledově pokusím udělat nějaký tvarový prototyp za pomoci té NACA sweep knihovny.  Nemyslim si totiž, že je to nějak zjevně snažší řešení toho problému (Protože už jen kvůli tisku se tam bude muset nějak upravovat asi odtoková hrana a natáčet na ní).  Zkusím to nejdřív ověřit.

slimonslimon, 2020-10-05:
@Elizaa62 co nějaká schůzka a tam to probrat? (mimochodem, všechno nejlepší k svátku)

Zabývám se také myšlenkou, kterou naznačil a navrhnul @roman-dvorak, a to je konstkrukce pomocí uhlíkových tyček. Návrh jejich upnutí by totiž byl poměrně jednoduchý. v podstatě to co psal @kaklik tady:
https://github.com/ThunderFly-aerospace/TF-G2/issues/9#issuecomment-647020654

Elizaa62, 2020-10-07:
Pokusila jsem se o aktualizaci pomocí návrhu sesikmene_nohy_2.. přidala jsem i prořezy
[image]

kaklik, 2020-10-09:
[image]

Tvar podvozku již funguje správně, ale aby podvozková noha šla použít, tak stále vyžaduje některé úpravy. 

- [ ] Tvar by měl být dutý, aby měla nízkou hmotnost,  tak jak to bylo před úpravou rozšířením
- [ ] Odlehčovací trojúhelníku musí měnit podél délky nohy svojí výšku, aby tloušťka všech stěn zůstávala zachována
- [ ] Podvozkovou nohu je potřeba alespoň z jedné strany sploštit, aby ji bylo možné tisknout položenou na podložce.  Tento krok zahrnuje otočení zužující se strany podvozkové nohy rovnoběžně s podložkou.

Elizaa62, 2020-10-20:
1. Dutost - parametr "zmenseni"
2. změna trojúhelníku - to nevím, jak udělat
3. splošení - to si nedokážu představit

kaklik, 2020-10-22:
>     1. Dutost - parametr "zmenseni"

Jo, to se docela povedlo! I jsem si vyzkoušel, že to funguje a dělá to co potřebujem. 

Na začátek jsem tam připsal do komentářů návrhy na nové hodnoty, neboť ta patka za k terou se podvozková noha přimontovává k vírníku se musí zvětšit. Když jsem to ale udělal, tak se to trochu rozsypalo. Takže jsem ty hodnoty vrátil zpátky a ty požadované jsem nechal v kometářích.

> 
>     2. změna trojúhelníku - to nevím, jak udělat

Zkusil jsem ti do jednoho toho for cyklu dopsat příklad, jak by to zhruba mohlo jít vyřešit. 
Mění to parametr průměru, pomocí kterého se vytváří ty trojuhelníky. Ty parametry, které jsem tam dopsal jsou jen pokus, správnější by zřejmě bylo je nějak chytřeji spočítat. 
Teď ale zřejmě nevadí že budou odhadnuté. (jen by to bylo potřeba udělat rovnou na tu rozšířenou nohu, viz výše.)

> 
>     3. splošení - to si nedokážu představit

Tady jsem udělal trochu obrázek toho, jak se to má tisknout.  Čevnený kvádr odpovídá tiskové podložce na kterou je ta podvozková noha položena při tisku. Ve tvém prvním návrhu, byla tahle strana zploštěná (uříznutá kovádrem), aby to na podložce drželo. To je myslím stále rozumný způsob jak to vyřešit. V tomhle návrhu ale tehlenhle uřezávací kvádr zatím chybí. V důsledku toho pak na podložku vychází ostrá hrana, která na tiskové podložce nebude držet. 

[image]

Tenhle obrázek ukazuje, že je podvozková noha navíc trošku otočená, aby bylo na podložku sedělo. Tohle otočení bych navrhoval dělat až v [tomhle tiskovém souboru](https://github.com/ThunderFly-aerospace/TF-G2/blob/debbf4c73f28c2ae5a0a2d58d68ac2b3ddb91d1a/CAD/src/printplate/printplate_888_2002.scad#L6-L7). Zatím tam jsou opět nějaké odhadnuté hodnoty.

kaklik, 2020-12-28:
Pokusím se shrnout aktuální stav, aby tohle issue bylo zas trochu přehledé

* Máme a aktuálně používáme [tuto verzi](https://github.com/ThunderFly-aerospace/TF-G2/issues/9#issuecomment-696098022) tato verze by potřobavala upravit pro větší odolnost proti vylomení při přistání. Vylomení je způsobeno předozadním zkroucením podvozku.  Existuje předpoklad, že k vyřešení tohoto problému by stačilo podvozkovou nohu rozšířit směrem ke trupu vírníku - na této úpravě už @Elizaa62  a @slimonslimon odvedli část práce.  Stav je viditelný z postů výše.
* V mezičase vznikly návrhy na dvě další konstrukční řešení podvozku, které ale nebyly zatím realizovány v žádné podobě. 
    1. [Konstrukce založená na aerodynamickém profilu ](https://github.com/ThunderFly-aerospace/TF-G2/issues/9#issuecomment-702710622) - jde spíše o řešení konstrukčních obtíží s návrhem podvozku v OpenSCADu než o významně odlišnou realizaci.
    2. [Konstrukční řešení využivající pružinové dráty](https://github.com/ThunderFly-aerospace/TF-G2/issues/9#issuecomment-647020654) - zde by pružnost vytištěných dílů byla nahrazena absorbcí energie v pružinových drátech, které by byly zalepeny do vytištěných dílů. 
    3. Kostrukční řešení využívající uhlíkové tyčky - v tomto případě je konstrukce velmi podobná jako v případě pružinových drátů, akorát vzhledem k tomu, že uhlíkový profil není schopen absorbovat významné množství energie svým ohybem, tak by uložení trubek podvozku muselo být otočné a absorbce energie řešena nějakým externím absorberem (guma, píst, tlumič). Toto řešení se technicky velmi podobá aktuálnímu podvozku na TF-G1.

kaklik, 2021-01-13:
@JanKott  To zplacatění, kvůli tisku, které je v modelu aktuálně implementováno: 

[image]

Způsobuje, že na straně podvozku u podložky, není prakticky žádný materiál.  Myslím si, že by bylo potřeba, aby zde byla téměř kontinuální tloušťka materiálu, aby tisková vlákna byla spojitá přes celou délku podvozkové nohy.

JanKott, 2021-01-20:
Vrátil jsem díru přes celou patku pro šroub kolečka.
Zmenšil jsem trojúhelníky aby zůstalo více materiálu na horním a spodním (z pohledu tisku) okraji podvozku.
Ořez pro položení na podložku jsem zrcadlil i na druhou stranu nohy aby byla symetrická.

Otočení trojúhelníků jsem zatim neřešil. Když je jen pootočim tak se mi ten problém s tloušťkou jen přesune jinam.

kaklik, 2021-01-21:
Pokusil jsem se to vytisknout a skončil jsem na tomto problému: 

[image]

Je to způsobené tím, že současný [print-plate](https://github.com/ThunderFly-aerospace/TF-G2/blob/master/CAD/src/printplate/printplate_888_2002.scad) neřeší rotaci vůči podložce. Podvozková noha se tak tiskne do vzduchu. 

Druhá věc pak je, že současné uchycení kolečka je myslím zbytečně hluboké a bylo by nutné použít zbytečně dlouhý šroub. Myslím že je dostatečné, aby uchycení bylo stejně hluboké, jako je tloušťka příhradové části s trojůhelníky.

[image]

JanKott, 2021-01-22:
Otočil jsem díl aby seděl na podložku a zmenšil podvozkový blok.

kaklik, 2021-02-02:
Při tisku se mi stala taková nepříjemnost, že z podvozku vzniklo něco jako siamská dvojčata. :D 

[image]

Tiskovou soustavu jsem ale opravil a už se tiskne nová verze i se vzdálenějšími otvory pro šrouby.  Nicméně si myslím, že tahle kostrukce podvozkové nohy bude už úplně v pohodě, ta tuhost je výtečná do všech směrů.  :)

kaklik, 2021-02-17:
Tak řešení podvozku se při posledních letech docela osvědčilo. :) 

Při "přistání" sice došlo k utržení podvozkové nohy od kostky, která podvozkovou nohu drží na šroubech. Ale řešení tohoto detailu jednak už není přímo předmětem tohoto issue a potom tato přistání asi nelze považovat za úplně relevatní k tomu co podvozek má vydržet.

closed 2021-02-17 by kaklik

slimonslimon, 2021-02-18:
dobrá práce @JanKott