ThunderFly-aerospace/TF-G2 #16: Doplnění airspeed senzoru

kaklik, 2020-04-26:
Na vírník je potřeba přidat [airspeed senzor](https://github.com/ThunderFly-aerospace/TFAIRSPD01).

Uchycen by měl být na horní části pylonu pod rotorovou hlavou. viz issue: https://github.com/ThunderFly-aerospace/TF-G2/issues/7

[image]

@roman-dvorak  Máš nějakou konkrétní představu o umístění?

roman-dvorak, 2020-04-26:
Úplně původně zamýšlené místo bylo pod tím středním rámečkem, kde jsou již připravené montážní otvory. Ale ten je dost nízko (původně měl být umístěný výš). Tak je možné ho nějak připevnit až přimo pod rotor. Do toho horního rámečku. 

[image]

kaklik, 2020-06-05:
Myslím že mě napadlo jak zjednodušit úpravu z issue https://github.com/ThunderFly-aerospace/TFASPD01/issues/1
Tak aby snimač bylo možné umístit úplně nejblíže rotorové hlavě. Venturiho snímač by tak nově byl takováto krabička:

[image]

Zaoblené profily jsou ve skutečnosti nějaké NACA profily a jsou symetrické dva proti sobě.  Předpokládám, že by se celá tahle hranatá věc tiskla najednou, takže to může být i nějaká přepážka sloužící k vyztužení.  
Je myslím zatím i jedno, jestli bude orientována horizontálně, nebo vertikálně. 
Plošný spoj se snímači by se opět přimáčkl do kanálků z vnější strany a držel by tam tím že se celý snímač k něčemu přišroubuje.

kaklik, 2020-06-18:
@ChroustJan  navrhuje přes PCB s elektronikou udělat zasouvací víčko. Výhoda by byla, že se pak celý snímač dá vyndat jako samostatný celek a nehrozí že někde na louce vypadnou těsnění, nebo PCB.

kaklik, 2020-09-30:
@roman-dvorak  udělal první prototyp venturiho poličky. Zatím je poměrně masivní. 

[image]

[image]

Chyby, které při návrhu vznikly jsou způsobeny nedokonalým převzetím konstrukce z anemometru. 

- [ ] Plošný spoj se neopírá o přívodní konektor, ale je výše. Díky čemuž jej nelze dobře upevnit vůči vyústění trubiček
- [ ] Vyústění trubiček neobsahuje zúžení, které je důležité pro vložení těsnícího o-kroužku. 
- [ ] celá konstrukce je zřejmě zbytečně masivní.

roman-dvorak, 2020-10-01:
> Plošný spoj se neopírá o přívodní konektor, ale je výše. Díky čemuž jej nelze dobře upevnit vůči vyústění trubiček

aha.. tohle jsem nepochopil správně.. takže teď bude konektor ještě výše.

kaklik, 2020-10-01:
Dnes jsem přigumičkoval snímač k nově vytištěné zmenšené variantě. 

[image]

[image]

Tohle se stane, když do toho fouknu: 

    0395; 2020-10-01T20:57:45.129444; Dp: +0.00 [Pa]; T: 32.774 [degC]; MAG_HDG: +63.04; SPD_W_DP: +0.00 [km/h]
    0396; 2020-10-01T20:57:45.183406; Dp: +0.00 [Pa]; T: 32.966 [degC]; MAG_HDG: +62.98; SPD_W_DP: +0.00 [km/h]
    0397; 2020-10-01T20:57:45.237408; Dp: +0.00 [Pa]; T: 33.110 [degC]; MAG_HDG: +62.77; SPD_W_DP: +0.00 [km/h]
    0398; 2020-10-01T20:57:45.291341; Dp: +0.00 [Pa]; T: 33.062 [degC]; MAG_HDG: +62.71; SPD_W_DP: +0.00 [km/h]
    0399; 2020-10-01T20:57:45.345361; Dp: -24.00 [Pa]; T: 33.062 [degC]; MAG_HDG: +62.59; SPD_W_DP: +1.76 [km/h]
    0400; 2020-10-01T20:57:45.399374; Dp: -34.15 [Pa]; T: 33.110 [degC]; MAG_HDG: +62.70; SPD_W_DP: +2.10 [km/h]
    0401; 2020-10-01T20:57:45.453341; Dp: -38.95 [Pa]; T: 33.062 [degC]; MAG_HDG: +62.50; SPD_W_DP: +2.25 [km/h]
    0402; 2020-10-01T20:57:45.507374; Dp: -55.35 [Pa]; T: 33.253 [degC]; MAG_HDG: +62.02; SPD_W_DP: +2.68 [km/h]
    0403; 2020-10-01T20:57:45.561325; Dp: -64.85 [Pa]; T: 33.157 [degC]; MAG_HDG: +61.83; SPD_W_DP: +2.90 [km/h]
    0404; 2020-10-01T20:57:45.615369; Dp: -65.85 [Pa]; T: 33.062 [degC]; MAG_HDG: +61.66; SPD_W_DP: +2.92 [km/h]
    0405; 2020-10-01T20:57:45.669336; Dp: -74.90 [Pa]; T: 33.062 [degC]; MAG_HDG: +61.41; SPD_W_DP: +3.12 [km/h]
    0406; 2020-10-01T20:57:45.723355; Dp: -94.05 [Pa]; T: 33.157 [degC]; MAG_HDG: +61.27; SPD_W_DP: +3.49 [km/h]
    0407; 2020-10-01T20:57:45.777320; Dp: -67.15 [Pa]; T: 33.062 [degC]; MAG_HDG: +61.31; SPD_W_DP: +2.95 [km/h]
    0408; 2020-10-01T20:57:45.831347; Dp: -75.90 [Pa]; T: 33.253 [degC]; MAG_HDG: +60.68; SPD_W_DP: +3.14 [km/h]
    0409; 2020-10-01T20:57:45.885315; Dp: -71.70 [Pa]; T: 33.110 [degC]; MAG_HDG: +60.26; SPD_W_DP: +3.05 [km/h]
    0410; 2020-10-01T20:57:45.939326; Dp: -86.30 [Pa]; T: 33.110 [degC]; MAG_HDG: +59.96; SPD_W_DP: +3.34 [km/h]
    0411; 2020-10-01T20:57:45.993290; Dp: -82.25 [Pa]; T: 33.110 [degC]; MAG_HDG: +60.01; SPD_W_DP: +3.26 [km/h]
    0412; 2020-10-01T20:57:46.047329; Dp: -58.25 [Pa]; T: 33.110 [degC]; MAG_HDG: +60.58; SPD_W_DP: +2.75 [km/h]
    0413; 2020-10-01T20:57:46.101322; Dp: -111.90 [Pa]; T: 33.110 [degC]; MAG_HDG: +60.22; SPD_W_DP: +3.81 [km/h]
    0414; 2020-10-01T20:57:46.155342; Dp: -115.60 [Pa]; T: 33.062 [degC]; MAG_HDG: +60.43; SPD_W_DP: +3.87 [km/h]
    0415; 2020-10-01T20:57:46.209296; Dp: -123.40 [Pa]; T: 33.110 [degC]; MAG_HDG: +60.19; SPD_W_DP: +4.00 [km/h]
    0416; 2020-10-01T20:57:46.263326; Dp: -103.10 [Pa]; T: 33.349 [degC]; MAG_HDG: +60.30; SPD_W_DP: +3.66 [km/h]
    0417; 2020-10-01T20:57:46.317298; Dp: -74.25 [Pa]; T: 33.110 [degC]; MAG_HDG: +60.11; SPD_W_DP: +3.10 [km/h]
    0418; 2020-10-01T20:57:46.371327; Dp: -96.85 [Pa]; T: 33.110 [degC]; MAG_HDG: +60.13; SPD_W_DP: +3.54 [km/h]
    0419; 2020-10-01T20:57:46.425293; Dp: -72.90 [Pa]; T: 33.110 [degC]; MAG_HDG: +60.21; SPD_W_DP: +3.07 [km/h]
    0420; 2020-10-01T20:57:46.479305; Dp: -80.70 [Pa]; T: 33.110 [degC]; MAG_HDG: +60.26; SPD_W_DP: +3.23 [km/h]
    0421; 2020-10-01T20:57:46.533287; Dp: -22.30 [Pa]; T: 33.110 [degC]; MAG_HDG: +60.34; SPD_W_DP: +1.70 [km/h]
    0422; 2020-10-01T20:57:46.587304; Dp: -0.40 [Pa]; T: 33.110 [degC]; MAG_HDG: +60.25; SPD_W_DP: +0.23 [km/h]
    0423; 2020-10-01T20:57:46.641271; Dp: -0.10 [Pa]; T: 33.110 [degC]; MAG_HDG: +60.66; SPD_W_DP: +0.11 [km/h]
    0424; 2020-10-01T20:57:46.695276; Dp: +0.00 [Pa]; T: 33.110 [degC]; MAG_HDG: +61.02; SPD_W_DP: +0.00 [km/h]
    0425; 2020-10-01T20:57:46.749256; Dp: +0.00 [Pa]; T: 33.445 [degC]; MAG_HDG: +61.12; SPD_W_DP: +0.00 [km/h]
    0426; 2020-10-01T20:57:46.803318; Dp: +0.05 [Pa]; T: 33.157 [degC]; MAG_HDG: +60.91; SPD_W_DP: -0.08 [km/h]
    0427; 2020-10-01T20:57:46.857261; Dp: +0.00 [Pa]; T: 33.157 [degC]; MAG_HDG: +60.97; SPD_W_DP: +0.00 [km/h]
    0428; 2020-10-01T20:57:46.911293; Dp: +0.05 [Pa]; T: 33.349 [degC]; MAG_HDG: +61.18; SPD_W_DP: -0.08 [km/h]
    0429; 2020-10-01T20:57:46.965244; Dp: +0.00 [Pa]; T: 33.253 [degC]; MAG_HDG: +61.27; SPD_W_DP: +0.00 [km/h]

To stejné jsem zkoušel i s variantou s NACA profilem

[image]

[image]

Když do ní fouknu tak to dopadne takhle

    0650; 2020-10-01T21:20:30.821502; Dp: +0.00 [Pa]; T: 34.739 [degC]; MAG_HDG: +82.74; SPD_W_DP: +0.00 [km/h]
    0651; 2020-10-01T21:20:30.875542; Dp: +0.00 [Pa]; T: 34.931 [degC]; MAG_HDG: +82.87; SPD_W_DP: +0.00 [km/h]
    0652; 2020-10-01T21:20:30.929516; Dp: +0.00 [Pa]; T: 35.074 [degC]; MAG_HDG: +82.64; SPD_W_DP: +0.00 [km/h]
    0653; 2020-10-01T21:20:30.983486; Dp: +0.00 [Pa]; T: 34.739 [degC]; MAG_HDG: +82.72; SPD_W_DP: +0.00 [km/h]
    0654; 2020-10-01T21:20:31.037518; Dp: +0.00 [Pa]; T: 34.931 [degC]; MAG_HDG: +83.25; SPD_W_DP: +0.00 [km/h]
    0655; 2020-10-01T21:20:31.091495; Dp: +0.00 [Pa]; T: 35.026 [degC]; MAG_HDG: +84.16; SPD_W_DP: +0.00 [km/h]
    0656; 2020-10-01T21:20:31.145503; Dp: +0.00 [Pa]; T: 35.074 [degC]; MAG_HDG: +84.07; SPD_W_DP: +0.00 [km/h]
    0657; 2020-10-01T21:20:31.199470; Dp: -51.95 [Pa]; T: 35.074 [degC]; MAG_HDG: +84.48; SPD_W_DP: +2.59 [km/h]
    0658; 2020-10-01T21:20:31.253509; Dp: -62.55 [Pa]; T: 34.931 [degC]; MAG_HDG: +84.48; SPD_W_DP: +2.85 [km/h]
    0659; 2020-10-01T21:20:31.307490; Dp: -167.35 [Pa]; T: 34.691 [degC]; MAG_HDG: +83.72; SPD_W_DP: +4.66 [km/h]
    0660; 2020-10-01T21:20:31.361531; Dp: -190.20 [Pa]; T: 34.883 [degC]; MAG_HDG: +83.08; SPD_W_DP: +4.96 [km/h]
[... cut ...]
    0683; 2020-10-01T21:20:32.603450; Dp: +0.05 [Pa]; T: 35.026 [degC]; MAG_HDG: +85.00; SPD_W_DP: -0.08 [km/h]
    0684; 2020-10-01T21:20:32.657477; Dp: +0.00 [Pa]; T: 35.170 [degC]; MAG_HDG: +88.92; SPD_W_DP: +0.00 [km/h]

Je vidět že obě varianty měří celkem dobře. Takhle se zdá, že varianta s NACA profilem je citlivější. Měření ale nepovažuji za příliš průkazné, protože jsem jednak přendaval čidlo z jedné varianty na druhou. Navíc fouknutí není zrovna dobře definovaný zdroj proudění vzduchu.

kaklik, 2020-10-01:
Tak jsem teď našel ještě druhý senzor a stejným způsobem ho připevnil na jeden z testovacích výtisků. Nyní prohazuji jen připojovací konektor. Výsledek je, že varianta s NACA profilem skutečně opakovatelně dává vyšší hodnoty tlakového spádu.

slimonslimon, 2020-10-02:
Dobrá to vypadá zajímavě. Takže teď je potřeba změřit závislost toho rozdílu na rychlosti proudějí.

Trochu "problém", který zatím předpokládám nebude řešit je závislost na veliskosti té krabičky a šířce štěrbiny.

Předpokládám, že naměříme nějakou křivku závislosti a z té uděláme "lookup tabulku" do PX4 funkcí na převod z diff preasure do airspeed.

kaklik, 2020-10-02:
> Dobrá to vypadá zajímavě. Takže teď je potřeba změřit závislost toho rozdílu na rychlosti proudějí.
>
> Trochu "problém", který zatím předpokládám nebude řešit je závislost na veliskosti té krabičky a [...]

Já bych se ohledně té implementace držel spíš toho současného způsobu implementace, který v PX4 je. Kdy to počítají polynomem. Respektive zkorigovanou rovnicí pro pitotku.  My možná můžeme vyjít z výpočtů okolo venturiho trubky: https://www.efunda.com/formulae/fluids/venturi_flowmeter.cfm 

Že to bude závislé na parametrech modelu je jasné, ale to by měly doufám postihovat ty koeficienty v té rovnici.   Ono jinak není úplně důležitý, aby to měřilo úplně přesně správnou rychlost. Jde spíš o to, aby jí to měřilo vždycky stejně.

kaklik, 2020-10-06:
Vyzkoušel jsem nově navržené díly.

[image]

Ten s menším průřezem: 

    0248; 2020-10-06T12:02:07.139813; Dp: +0.00 [Pa]; T: 27.455 [degC]; MAG_HDG: +356.53; SPD_W_DP: +0.00 [km/h]
    0249; 2020-10-06T12:02:07.193830; Dp: +0.00 [Pa]; T: 27.503 [degC]; MAG_HDG: +356.34; SPD_W_DP: +0.00 [km/h]
    0250; 2020-10-06T12:02:07.247810; Dp: +0.00 [Pa]; T: 27.503 [degC]; MAG_HDG: +356.20; SPD_W_DP: +0.00 [km/h]
    0251; 2020-10-06T12:02:07.301841; Dp: -26.55 [Pa]; T: 27.503 [degC]; MAG_HDG: +356.20; SPD_W_DP: +1.85 [km/h]
    0252; 2020-10-06T12:02:07.355808; Dp: -150.20 [Pa]; T: 27.790 [degC]; MAG_HDG: +356.16; SPD_W_DP: +4.41 [km/h]
    0253; 2020-10-06T12:02:07.409821; Dp: -123.00 [Pa]; T: 27.550 [degC]; MAG_HDG: +356.17; SPD_W_DP: +3.99 [km/h]
    0254; 2020-10-06T12:02:07.463804; Dp: -410.50 [Pa]; T: 27.311 [degC]; MAG_HDG: +356.11; SPD_W_DP: +7.29 [km/h]
    0255; 2020-10-06T12:02:07.517821; Dp: -536.10 [Pa]; T: 27.359 [degC]; MAG_HDG: +355.94; SPD_W_DP: +8.33 [km/h]
    0256; 2020-10-06T12:02:07.571801; Dp: -592.35 [Pa]; T: 27.455 [degC]; MAG_HDG: +355.85; SPD_W_DP: +8.76 [km/h]
    0257; 2020-10-06T12:02:07.625815; Dp: -592.45 [Pa]; T: 27.694 [degC]; MAG_HDG: +355.66; SPD_W_DP: +8.76 [km/h]
    0258; 2020-10-06T12:02:07.679824; Dp: -616.80 [Pa]; T: 27.311 [degC]; MAG_HDG: +355.40; SPD_W_DP: +8.94 [km/h]
    0259; 2020-10-06T12:02:07.733831; Dp: -498.60 [Pa]; T: 27.503 [degC]; MAG_HDG: +355.25; SPD_W_DP: +8.04 [km/h]
    0260; 2020-10-06T12:02:07.787824; Dp: -659.45 [Pa]; T: 27.550 [degC]; MAG_HDG: +355.13; SPD_W_DP: +9.24 [km/h]
    0261; 2020-10-06T12:02:07.841794; Dp: -710.35 [Pa]; T: 27.455 [degC]; MAG_HDG: +355.20; SPD_W_DP: +9.59 [km/h]
    0262; 2020-10-06T12:02:07.895788; Dp: -774.80 [Pa]; T: 27.455 [degC]; MAG_HDG: +354.97; SPD_W_DP: +10.02 [km/h]
    0263; 2020-10-06T12:02:07.949773; Dp: -811.10 [Pa]; T: 27.646 [degC]; MAG_HDG: +354.90; SPD_W_DP: +10.25 [km/h]
    0264; 2020-10-06T12:02:08.003820; Dp: -771.00 [Pa]; T: 27.503 [degC]; MAG_HDG: +354.75; SPD_W_DP: +10.00 [km/h]
    0265; 2020-10-06T12:02:08.057786; Dp: -730.80 [Pa]; T: 27.550 [degC]; MAG_HDG: +354.70; SPD_W_DP: +9.73 [km/h]
    0266; 2020-10-06T12:02:08.111801; Dp: -691.65 [Pa]; T: 27.503 [degC]; MAG_HDG: +354.57; SPD_W_DP: +9.47 [km/h]
    0267; 2020-10-06T12:02:08.165812; Dp: -759.20 [Pa]; T: 27.694 [degC]; MAG_HDG: +354.22; SPD_W_DP: +9.92 [km/h]
    0268; 2020-10-06T12:02:08.219802; Dp: -770.30 [Pa]; T: 27.503 [degC]; MAG_HDG: +353.91; SPD_W_DP: +9.99 [km/h]
    0269; 2020-10-06T12:02:08.273791; Dp: -692.95 [Pa]; T: 27.503 [degC]; MAG_HDG: +353.78; SPD_W_DP: +9.48 [km/h]
    0270; 2020-10-06T12:02:08.327779; Dp: -773.15 [Pa]; T: 27.503 [degC]; MAG_HDG: +353.63; SPD_W_DP: +10.01 [km/h]
    0271; 2020-10-06T12:02:08.381828; Dp: -706.05 [Pa]; T: 27.455 [degC]; MAG_HDG: +353.48; SPD_W_DP: +9.56 [km/h]
    0272; 2020-10-06T12:02:08.435777; Dp: -761.45 [Pa]; T: 27.503 [degC]; MAG_HDG: +353.49; SPD_W_DP: +9.93 [km/h]
    0273; 2020-10-06T12:02:08.489772; Dp: -576.45 [Pa]; T: 27.646 [degC]; MAG_HDG: +353.65; SPD_W_DP: +8.64 [km/h]
    0274; 2020-10-06T12:02:08.543760; Dp: -96.35 [Pa]; T: 27.503 [degC]; MAG_HDG: +353.58; SPD_W_DP: +3.53 [km/h]
    0275; 2020-10-06T12:02:08.597806; Dp: -7.05 [Pa]; T: 27.503 [degC]; MAG_HDG: +353.36; SPD_W_DP: +0.96 [km/h]
    0276; 2020-10-06T12:02:08.651771; Dp: -0.50 [Pa]; T: 27.455 [degC]; MAG_HDG: +353.33; SPD_W_DP: +0.25 [km/h]
    0277; 2020-10-06T12:02:08.705786; Dp: -0.10 [Pa]; T: 27.455 [degC]; MAG_HDG: +353.55; SPD_W_DP: +0.11 [km/h]
    0278; 2020-10-06T12:02:08.759762; Dp: -0.05 [Pa]; T: 27.790 [degC]; MAG_HDG: +353.70; SPD_W_DP: +0.08 [km/h]
    0279; 2020-10-06T12:02:08.813775; Dp: +0.00 [Pa]; T: 27.311 [degC]; MAG_HDG: +353.94; SPD_W_DP: +0.00 [km/h]
    0280; 2020-10-06T12:02:08.867762; Dp: +0.00 [Pa]; T: 27.503 [degC]; MAG_HDG: +353.93; SPD_W_DP: +0.00 [km/h]
    0281; 2020-10-06T12:02:08.921758; Dp: +0.00 [Pa]; T: 27.646 [degC]; MAG_HDG: +353.89; SPD_W_DP: +0.00 [km/h]
    0282; 2020-10-06T12:02:08.975795; Dp: +0.00 [Pa]; T: 27.646 [degC]; MAG_HDG: +354.09; SPD_W_DP: +0.00 [km/h]
    0283; 2020-10-06T12:02:09.029759; Dp: +0.00 [Pa]; T: 27.455 [degC]; MAG_HDG: +354.03; SPD_W_DP: +0.00 [km/h]

S větší škvírou: 

    0372; 2020-10-06T12:01:04.614593; Dp: +0.00 [Pa]; T: 27.838 [degC]; MAG_HDG: +204.13; SPD_W_DP: +0.00 [km/h]
    0373; 2020-10-06T12:01:04.668620; Dp: +0.00 [Pa]; T: 27.550 [degC]; MAG_HDG: +204.24; SPD_W_DP: +0.00 [km/h]
    0374; 2020-10-06T12:01:04.722614; Dp: +0.00 [Pa]; T: 27.503 [degC]; MAG_HDG: +204.43; SPD_W_DP: +0.00 [km/h]
    0375; 2020-10-06T12:01:04.776587; Dp: -24.75 [Pa]; T: 27.838 [degC]; MAG_HDG: +204.60; SPD_W_DP: +1.79 [km/h]
    0376; 2020-10-06T12:01:04.830631; Dp: -29.40 [Pa]; T: 27.982 [degC]; MAG_HDG: +204.95; SPD_W_DP: +1.95 [km/h]
    0377; 2020-10-06T12:01:04.884589; Dp: -133.55 [Pa]; T: 27.694 [degC]; MAG_HDG: +205.00; SPD_W_DP: +4.16 [km/h]
    0378; 2020-10-06T12:01:04.938601; Dp: -314.85 [Pa]; T: 27.790 [degC]; MAG_HDG: +205.36; SPD_W_DP: +6.39 [km/h]
    0379; 2020-10-06T12:01:04.992581; Dp: -387.90 [Pa]; T: 27.646 [degC]; MAG_HDG: +205.44; SPD_W_DP: +7.09 [km/h]
    0380; 2020-10-06T12:01:05.046589; Dp: -458.65 [Pa]; T: 27.838 [degC]; MAG_HDG: +205.77; SPD_W_DP: +7.71 [km/h]
    0381; 2020-10-06T12:01:05.100580; Dp: -315.65 [Pa]; T: 27.790 [degC]; MAG_HDG: +206.01; SPD_W_DP: +6.40 [km/h]
    0382; 2020-10-06T12:01:05.154613; Dp: -343.50 [Pa]; T: 27.838 [degC]; MAG_HDG: +206.05; SPD_W_DP: +6.67 [km/h]
[... cut ...]
    0397; 2020-10-06T12:01:05.964582; Dp: -0.70 [Pa]; T: 27.982 [degC]; MAG_HDG: +204.83; SPD_W_DP: +0.30 [km/h]
    0398; 2020-10-06T12:01:06.018563; Dp: -0.10 [Pa]; T: 27.838 [degC]; MAG_HDG: +204.96; SPD_W_DP: +0.11 [km/h]
    0399; 2020-10-06T12:01:06.072573; Dp: -0.05 [Pa]; T: 27.790 [degC]; MAG_HDG: +204.97; SPD_W_DP: +0.08 [km/h]
    0400; 2020-10-06T12:01:06.126581; Dp: +0.00 [Pa]; T: 27.886 [degC]; MAG_HDG: +204.64; SPD_W_DP: +0.00 [km/h]
    0401; 2020-10-06T12:01:06.180529; Dp: +0.00 [Pa]; T: 27.646 [degC]; MAG_HDG: +213.57; SPD_W_DP: +0.00 [km/h]

kaklik, 2020-10-09:
Možná by šlo odstranit ty tiskové podpory tím, že by se jen zvýšil úhel těch uší směrem od podložky. Jsou nyní totiž výrazně užší, než je šířka pylonu. 

[image]

kaklik, 2020-10-27:
@JanKott  na fotografiích výše můžeš vidět, jakým způsobem (gumičkou) je tam aktuálně připevněno měřící PCB. Jde o identický plošný spoj, jako je ve windgauge.

V modelu je příprava na šrouby, které by měly sloužit pro upevnění krytu. 
Kryt ale zatím nikdo neudělal, protože stačila gumička. Kryt by měl být udělaný tak, aby vodiče šlo vyvést po jedné straně pylonu a nevadily průtoku vzduchu.

kaklik, 2020-10-30:
@roman-dvorak  Připravil víčko, které slouží k upevnění elektroniky. 

Zatím ale nesplňuje požadavek na nízký aerodynamický odpor, protože v přední části má rovnou plochu. Myslím že by to ale mohlo jít vyřešit tím, že se přední část upraví tak, aby plynule navazovala na aerodynamický profil venturiho poličky.

Důvod proč to tak teď není je především v tom, že je potřeba vymyslet, jak takovou věc tisknout. (Aktuálně se víčko tiskne položené na spodní rovnou plochu)

[image]

[image]

[image]

roman-dvorak, 2020-11-02:
# Testování venturiho poličky

Venturiho poličku jsem připevnil na střechu auta. Cca 60 cm nad střechu. 
[image]

### Měření
Měření bylo provedeno pomocí [skriptu](https://github.com/ThunderFly-aerospace/TFASPDIMU01/blob/TFASPD01A/sw/TFASPD_logger.py). GPS byla připojena do Omnie s LTE/GPS přijímačem a připojena do GPSD. Externí anténa byla přimagnetovaná na střechu auta. 

Každé měření bylo provedeno na jiné cestě v normálním provozu. Proto podmínky byly dost nedefinované. Tento den však nefoukalo. Cílem bylo jet určité úseky s pomocí tempomatu (tedy konstantní rychlostí). 

### Legenda ke grafům

- První graf je graf groundspeed z GPS X naměřený tlak z differencionálního senzoru. 

- Druhý graf zobrazuje průběh měření. Modrý graf (levý sloupec) je GPS v km/h, Světle červený graf (pravý sloupec) je rozdíl tlaku v Pa. 

## NACA0040, d=20

[image]

[image]

## NACA0045, d=20

[image]

[image]

roman-dvorak, 2020-11-09:
Docela výhodné bude venturiho poličku montovat obráceně (PCB nahoře) a víčko implementovat do nejvyššího dílu pylonu. Polička bude k pylonu připevňena z horní strany pomocí dvojice šroubů. Tak, jak je to v poličce aktuálně implementováno. 

Toto vylepšení souvisí s jinou montáží serv #70

JanKott, 2020-11-10:
Můžete mi prosím nakreslit jaká je požadovaná změna?

roman-dvorak, 2020-11-10:
Moje představa je cca taková:

[image]

Asi ani není nutné, aby to byl přesně NACA profil. Víčko bude na anemometru připevněno z horní strany. Dvojicí šroubů po stranách v přední části anemometru. Cílem je anemometr co nejvíce vysunout dopředu. 

V co nejzadnější části, (kde to materiál dovolí) by měla být příčná díra skrz a v ní dvě kapsy na matky. o něco nopřeději (cca 5-10 mm) by měl být ďolík (odebraná polokoule) která bude sloužit k aretaci anemometr tak, aby nemohlo dojít k jeho rotaci okolo dvojice šroubů. Hloubka stačí cca 2 mm, průměr cca 3 mm. 

Mezi poličkou a víčkem by bylo dobré mít nějaký rantl, který zamezí natečení vody do senzoru. Zároveň by bylo dobré, aby bylo možné senzor tisknout na bok. (tj. aby vrstva tisku odpovídala proudnicím skrze poličku) Aktuálně se to tiskne kolmo na proudnice (na odtokovou hranu) 

Víčko jsem předpokládal, že se bude tisknout na náběžnou hranu.

kaklik, 2020-11-17:
Dneska jsem o tom mluvil s @JanKott. Zde v popisu není uvedeno, že ta dvojice šroubů, která víčko upevňuje by měla procházet rantlem, aby skrz šrouby nemohlo napršet přímo na elektroniku snímače. 

Ten způsob tisku bych preferoval na bok a to zejména z důvodu, že na náběžnou, nebo odtokovou hranu se špatně tiskne. Vliv zoubků na aerodynamiku profilu bude pro tenhle případ myslím zanedbatelný.

roman-dvorak, 2020-11-25:
@JanKott, nějaký progress? :)

JanKott, 2020-11-26:
No progress at all :-/

JanKott, 2020-12-09:
Víčko je upravené dle požadavků.

JanKott, 2020-12-09:
Zavřený asi neni dobrý stav.. Předávám k otestování.

kaklik, 2020-12-25:
Jsou vytištěny a složeny snímače připravené k otestování. 

[image]

slimonslimon, 2020-12-25:
Při příštím letu by měl startovat můj stroj osazen těmito 2 poličkami a pitotovou trubicí (opravenou po havarii). Airspeed bude měžen pitotovou trubicí, tlaky budou logovány.

kaklik, 2021-02-17:
Tohle issue mi přijde vlastně hotové, až na ne úplně jasně uzavřený software.

slimonslimon, 2021-02-18:
na což je issue tady (?): https://github.com/ThunderFly-aerospace/TFASPDIMU01/issues/3 tudíž navrhuji toto issue uzavřít.

roman-dvorak, 2021-02-18:
> na což je issue tady (?): ThunderFly-aerospace/TFASPDIMU01#3
> <https://github.com/ThunderFly-aerospace/TFASPDIMU01/issues/3> tudíž
> navrhuji toto issue uzavřít.

Z HW stránky tomu zase chybí víčko, které bude vypadat asi trochu jinak.

kaklik, 2021-02-18:
> Z HW stránky tomu zase chybí víčko, které bude vypadat asi trochu jinak.

Pokud píšeš, že k tomu chybí víčko a zároveň bude vypadat asi trochu jinak. Tak zřejmě máš nějakou představu o tom, jak to víčko má vypadat, což tu není popsaný. :)

kaklik, 2021-02-23:
Zde je potřeba ještě: 

- [ ] zapracovat víčko airspeed senzoru držící PCB [TFASPDIMU](https://github.com/ThunderFly-aerospace/TFASPDIMU01)
- [ ] Vyřešit umístění TFRPM viz issue: https://github.com/ThunderFly-aerospace/TF-G2/issues/76

Nakonec si myslím, že by bylo potřeba dát TFRPM z vrchní strany, aby bylo možné zkontrolovat funkci RPM snímače na vírníku stojícím na zemi a nenastal stejný problém jako je popsaný v issue https://github.com/ThunderFly-aerospace/TF-G2/issues/76 kdy je LED vidět jen ze spodní strany. 

Myslím že by to šlo udělat tak, že se TFASPDIMU dá pod TFRM tak, že oba PCB se budou dotýkat plochou stranou bez součástek. Přes ně by pak mohlo být přitažené víčko přišroubované samořeznými šrouby.

kaklik, 2021-09-03:
Nové uchycení senzoru i s jednoduchým víčkem na šrouby funguje docela dobře.

closed 2021-09-03 by kaklik