ThunderFly-aerospace/TF-G2 #18: Dokončit docs.px4 stránku pro Auto-G2

roman-dvorak, 2020-05-12:
Cílem je naučit se, jak s touto dokumentací zacházet. Jaké bude mít komunita požadavky atd..

Text je rozepsán zde: https://docs.google.com/document/d/1p8QPR6BAmqDs6eTo3oPDJKxsFIAMStuHuRvq7hzDEPY/edit?fbclid=IwAR0um9o1BWj3zY5KZpIYuSvzSpZiG_S9OVbog_OKz805KD17tEJQtICigdU#

Je to text pro tuto stránku: https://docs.px4.io/master/en/airframes/

kaklik, 2020-05-13:
Tohle je duplicitní k: https://github.com/ThunderFly-aerospace/Auto-G2/issues/24

kaklik, 2020-05-20:
Synchronizoval jsem text dokumentu s markdownem v dokumentaci. Přidal jsem i odkazy pro některé tišné díly. Nyní je potřeba vložit obrázky a nějaké úplně základní video.

kaklik, 2020-09-29:
Rozsah by zřejmě měl být obdobný této stránce: https://docs.px4.io/master/en/frames_vtol/vtol_tailsitter_caipiroshka_pixracer.html

kaklik, 2020-09-30:
Vytvořen pull-request: https://github.com/PX4/px4_user_guide/pull/819

kaklik, 2020-10-07:
Pull-request zamegován a dokumentace je k dispozici na: https://docs.px4.io/master/en/frames_autogyro/thunderfly_auto_g2.html

closed 2020-10-07 by kaklik