ThunderFly-aerospace/TF-G2 #21: Návod pro stavbu TF-G2

roman-dvorak, 2020-05-12:
Pro vytvoření návodu je potřeba splnit tyto podúkoly:

- [ ] Vygenerovat ikea obrázky
- [ ] Připravit seznam dílů
- [ ] Napsat text
- [ ] Přeložit text do AJ
- [ ] Ověřit správnost návodu (ideálně někym třetím) 
- [ ] Vytvořit celkové fotky
- [ ] Připravit TF šablonu

kaklik, 2020-05-12:
Myslím, že každý tenhle bod by reálně měl mít svoje vlastní issue a tady by na ně měly být linky. A nebo by to měl být spíš samostatný projekt pod TF-G2

roman-dvorak, 2020-05-13:
@kaklik samostatný projekt pro vytvoření dokumentů k TF-G2 jsem již vytvořil. Je tady: https://github.com/ThunderFly-aerospace/TF-G2/projects/1#card-38058176

Jednotlivé body tohoto issue jsem rozepsal na vlastní úkoly.

kaklik, 2020-05-13:
Pokud je ale tohle issue rozepsané na jednotlivé úkoly roztřiděné v projektu, tak je tohle issue tím uzavřené, ne?

closed 2020-05-13 by roman-dvorak