ThunderFly-aerospace/TF-G2 #28: Vygenerovat nové ikea obrázky

roman-dvorak, 2020-05-13:
- [] Obrázky dílů
- [] Vyřešit externí díly pro stavbu
- [] Obrázky částečných sestav pro skladbu

kaklik, 2022-02-19:
Aktuální situace je, že pro všechny obrázky chybí openscad soubory. Vzniklé obrázky byly udělané vložením stl z modelů v základním stavu (přímo vyrenderované ze zdrojových kódů) do blenderu. 

Víme však že to s blenderem lze dělat a i plně zautomatizovat. Chybí zdroj dat (ty sestavy) a pak nějaké nastavení správných pohledů.

kittlerm, 2022-02-24:
V OpenSCAD se využívá funkce [children](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/User-Defined_Functions_and_Modules#Children), použitá je např. v [rotorové hlavě](https://github.com/ThunderFly-aerospace/TF-G2/blob/4s/CAD/src/assembly/position/position_rotorhead.scad#L15)

Diskuze na jitsimeet proběhla 21.2.2022 a mým úkolem by mělo být: zkusit definovat způsob, jak by šlo generování vylepšit (automatizovat, parametrizovat). Zápis ze setkání ještě doplním.

kittlerm, 2022-02-24:
V meetingu bylo zmíněno např. použití jsonu k definici scény: https://github.com/Hairein/MKS_BlenderToolScripts

kittlerm, 2022-11-19:
Chybně jsem označil změnu v processoru3D https://github.com/ThunderFly-aerospace/processor3D/commit/141366f48f1438add8be4d034d5da6fc4ebcec1e
 tímto issue, ale je to jiné repo