ThunderFly-aerospace/TF-G2 #38: Výběr barevné kombinace komponent vírníku

kaklik, 2020-05-18:
Pro zlepšení estetické kvality designu je nutné, aby vírník měl vhodnou barevnou kombinaci.  Z technických důvodů je vhodné, aby počet barev byl co nejnižší. 

Kvůli zajištění viditelnosti je vhodné, aby baravená kombinace zahrnovala světlou (viditelná proti tmavému pozadí) a tmavou barvu (viditelná proti obloze). 

Barevná kombinace by měla vycházet z grafického manuálu TF.  

Tento úkol je zřejmě vhodné začít řešit až po jiných konstrukčních úpravách, jako je:

- #15
- #16
- #9

kaklik, 2020-06-01:
Použitelné barevné vzorky jsou tyto: 

[image]

roman-dvorak, 2020-06-01:
Zřejmě to stačí udělat na tomto připraveném modelu:
https://github.com/ThunderFly-aerospace/TF-G2/blob/master/CAD/blender/TF-G2.blend

ze kterého vznikl například tento obrázek
[image]

roman-dvorak, 2020-06-18:
Po diskuzi s @ChroustJan a @kaklik to vypadá, že bude potřeba mít 2 verze. Jednu hezkou, která bude možná v TF barvách (konkrétní odstín oranžové) a druhou, která bude optimalizovaná pro nejlepší viditelnost na obloze.

roman-dvorak, 2020-06-18:
Ukázky:

[image]

[image]

[image]
[image]

roman-dvorak, 2020-07-03:
@fluktuacia navrhla následující barevnou kombinaci. Konstrukční prvky (tj. spodní podložka, příhradové bočnice, polička GPS, polička akumulátoru, přepážka motoru, pylon kompletně, rotorová hlava a pohyblivá část směrovky budou šedé. 

Ostatní části (okrytování, výškovka, směrovka a rotor) budou mít nějakou jednu barvu. Primárně se uvažuje o modré. Ale je možné jakoukoliv jinou. To bude podle požadavků zákazníka nebo letových podmínek. 

Konce rotoru by měly být nabarvené barvou, která je kontrastní ke zbytku. To především ke zvýšení bezpečnosti. A zároveň to vytváří docela zajímavý efekt. 

[image]

kaklik, 2020-07-03:
Konce rotoru nabarvené barvou?  Nemělo by to spíše být rovnou vytištěné z jiné barvy?

roman-dvorak, 2020-07-04:
Jo. Ano vytištěné. ale z jiné barvy prostě.

closed 2020-11-24 by roman-dvorak