ThunderFly-aerospace/TF-G2 #39: Vnější kryty vírníku

kaklik, 2020-05-18:
Vnější kryty namontované na skelet vírníku by mohly zlepšit následující parametry: 

- Vylepšit stetický dojem 
- Přidat deformační zóny a tím zlepšit odolnost
- Snížit aerodynamický odpor 

Je pravděpodobné, že tyto kryty budou 3D tištěné jednoperimetrovým tiskem.  Výměna těchto krytů by měla být relativně snadná, neboť lze očekávat jejich časté poškození. 
Alternativním řešením konstrukce krytů je tvarová změna některých konstrukčních částí tak, aby byly součástí krytu, V některých případech tak může docházet ke zvýšení tuhosti přenosem sil přes kryt.

kaklik, 2020-06-14:
Některým částem vírníku by se dala zlepšit aerodynamika jen i úpravou tvaru. Například svislé bočnice pylonu by mohly mít aerodynamickou náběžnou a odtokovou hranu.

kaklik, 2020-06-16:
Návrh na konstrukční řešení krytů. 
Jednotlivé úleky jsou spojeny plochami.  Plochy na sebe navazují s mírným radiusem. 

[image]

[image]

[image]

kaklik, 2020-07-28:
Vyřešeno aktuálním návrhem oblých krytů.

kaklik, 2021-03-25:
@taury55  Zkusil vytvořit kryty z nánosované textilie:

[image]

Tento typ krytů je velmi perspektivní, ale je nutné řešení trochu překonstruovat, aby plátěný obal byl napnutý na 3D tištěnou konstrukci.

kaklik, 2021-08-23:
Zde je nový prototyp z PUL tkaniny.  Hlavními problémy je: 

- Detail postupu pro pylon - musí být nějak utěsněn
- Přístup chladícího vzduchu k motoru. Přední otvory zřejmě musí být nějak odkryté, aby mohl motor nasávat chladící vzduch. Zároveň ale zde musí být relativně pevně uchycen lem obalu.
- Na spodní straně jsou nyní na zapnutí použity cvoky. Lepším řešením asi bude použití voděodolného zipu.

[image]
[image]

kaklik, 2021-08-30:
Jak se ukázalo na fotografiích výše, tak použití druků pro zapínání spodní části vírníku je nevhodné. Lepší bude zde použít voděodolný zip. 
Zůstane tak zřejmě jen pouze poslední druk až úplně v zadní části, který bude bránit nechtěnému rozepínání zipu. 

[image]

kaklik, 2021-09-04:
Zde je úprava po zašití voděodolného zipu

[image]

[image]

Obal je tak hotový až na místo prostupu kabelů z rotorového pylonu do trupu.  Zde bych asi navrhoval nakreslit 3D tištěný převlek, kterým se otvor v textilním obalu zakryje.

roman-dvorak, 2022-06-06:
@llucka vytvořila novou verzi oblečku. 

Při návrhu střihu jsme se snažili vyřešit některé záležitosti, které jsem u předchozího oblečku považoval za problematické. A to především:

 * Těžko se přistupuje k avionice uvnitř trupu. 
 * Není vyřešen prostup pro kolečka
 * Lepší chlazení motoru
 * Vyřešení prostupu pro vzduch
 * Snadná výměna akumulátoru
 * Kompatibilita s platformou

Vyřešení těchto bodů sebou nese zmenšení odolnosti proti vodě. Především ze spodní strany. Ale myslím, že je stále velmi vysoká i na to, aby bylo možné létat ve velkém dešti. Pláštěnka je na trup zafixovaná pomocí suchých zipů. 

Utěsnění pod pylonem je realizované tak, že v látce jsou "knoflíkové" otvory, které jsou nastrčené nad silentbloky. A stažené šrouby. držící pavouka pilonu. Utěsnění okolo drátů je realizované děravou kapsou, která zajišťuje dostatečnou voděodolnost. 

Potisk byl vyroben metodou plotrového tisku v tiskárně [tomprint](https://www.tomprint.cz)

**Nedořešené záležitosti:**

V přední části u motoru bych viděl, že vznikne nějaký vytištěný kroužek, který se nastrčí na přední část a bude tam pláštěnku fixovat. 

A v okolí podvozkových nohou budou vytištěná takové 'U' rámečky, které se nastrčí na podvozkovou nohu a pláštěnka tak taktéž bude zafixovaná na místě. Díky tomu by i okraj pláštěnky měl být hezký. 

Dalším vylepšením by bylo, abychom z tohoto střihu udělali nějakým způsobem "parametrický" střih, který by se generoval sám a zároveň by dbal na vlastnosti látky. 

[image]

[image]

kaklik, 2022-06-06:
Bezva :)  Asi si to pak budu muset ještě prohlédnout v reálu, ale není ta přední část takhle zbytečně komplikovaná? Vidím tam našitý další díl a zároveň stále ty podélné švy, které měla i předchozí verze.  Přišlo by mi úspornější to buď celé řešit přidaným kusem látky (což asi nejde) a nebo mít jen podélné švy, které umožní vstupní část zůžit.

llucka, 2022-06-07:
Ve přední části tam nejsou stejné švy jako u předchozího střihu. 
Celkové zúžení přední části je díky přednímu kusu látky, který je na to tak střihnutý. Ty dva malé přední švy by tam ani být nemuseli (je to jenom malilinký kousek látky, který je přehnutý).