ThunderFly-aerospace/TF-G2 #43: Změřit poláru vírniku

kaklik, 2020-06-01:
Vírníku by bylo potřeba [změřit poláru](http://www.airspace.cz/akademie/rocnik/2009/07/polara.php), abychom věděli při jaké rychlosti můžeme očekávat minimální opadání. 

[image]

Potenciálním problémem celého takového měření je, že polára se může v průběhu letu měnit.  Podle aktuálních parametrů hustota vzduchu, ušpinění rotoru, a podobně. 

Pro létání v termice by tak bylo výhodné, kdyby měřící aplikace uměla fungovat i jako MPPT traceker, který by optimální rychlost klouzání neustále hledal.

kaklik, 2021-06-26:
Tohle issue by zřejmě nyní, když vírník nepřechází do spirály, už mohlo být řešitelné. :)

kaklik, 2021-08-13:
Zjištění z https://github.com/ThunderFly-aerospace/PX4Firmware/issues/35 zřejmě usnadnilo možný způsob řešení.

kaklik, 2021-11-15:
K tomuhle je zřejmě potřeba funční "airspeed mód". 
FYI @roman-dvorak

roman-dvorak, 2021-11-15:
Nevím, jestli by nebyl vhodnější nějaký automatický režim (např loiter) s variabilním airspeed.. Kdy by se pak našel konkrétní airspeed setpoint s minimálním příkonem. Zřejmě ale bude existovat více optimálních hodnot. Například optimální airspeed pro dosažení co největší doby  letu vs optimální airspeed pro největší délku trajektorie letu. 

O to se snaží tady: https://github.com/PX4/PX4-Autopilot/pull/18605 Tahle implementace má bohužel 3 rychlostní režimy (AIRSPEED_MIN/TRIM/MAX) a neumožňuje plynulou změnu. (Tato úprava zároveň má zvětšit robustnost ve větru) 

Nalezení optimální rychlosti by možná mělo být i předmětem auto-tunning algoritmů https://github.com/PX4/PX4-Autopilot/pull/18330.

kaklik, 2021-11-15:
> Nevím, jestli by nebyl vhodnější nějaký automatický režim (např loiter) s variabilním airspeed.. Kdy by se pak našel konkrétní airspeed setpoint s minimálním příkonem. Zřejmě ale bude existovat více optimálních hodnot. Například optimální airspeed pro dosažení co největší doby letu vs optimální airspeed pro největší délku trajektorie letu.
> 
> O to se snaží tady: [PX4/PX4-Autopilot#18605](https://github.com/PX4/PX4-Autopilot/pull/18605) Tahle implementace má bohužel 3 rychlostní režimy (AIRSPEED_MIN/TRIM/MAX) a neumožňuje plynulou změnu. (Tato úprava zároveň má zvětšit robustnost ve větru)
> 
> Nalezení optimální rychlosti by možná mělo být i předmětem auto-tunning algoritmů [PX4/PX4-Autopilot#18330](https://github.com/PX4/PX4-Autopilot/pull/18330).

Ten automatický režim souvisí s tím jakým způsobem se vypořádáme s větrem. Protože v loiter režimu bude potíž s tím že během úseků na té kružnici bude mít vítr různou rychlost a směr. 
Myslím si tak, že pokud by to měl být automatický režim měření, tak by musel obsahovat rovné přelety ve směru proti větru. 
V automatickém režimu letu na kružnici by to ale určitě šlo provozavat v kombinaci s motorem. Je to ale myslím jiný typ měření.

kaklik, 2022-01-13:
Myslím si, že tohle by bylo zřejmě nejlepší implementovat ve formě [Flight Task](https://docs.px4.io/master/en/concept/flight_tasks.html)  uvnitř nějakého vhodného režimu letu. Nejvhodnější mi aktuálně přijde "airspeed" pro hledání parametrů optimálních parametrů bezmotorového letu. A pak "hold" pro hledání optimálních parametrů motorového letu. 

Zřejmě by asi bylo možné, aby tento "MPPT optimizer" běžel vlastně pořád.  Protože nyní se třeba s airspeed zachází tak, že se letí AIRSPEED_TRIM, dokud regulátor nenarazí na nějaké limity. Jako třeba menší než minimální ground speed.  Předpokládá se tak, že AIRSPEED_TRIM je konstanta po celou dobu letu a je neměnnou charakteristikou UAV, což ale prakticky není pravda. 
Optimizer by pak mohl AIRSPEED_TRIM měnit buď dynamicky, nebo pokud je schopen získat nějaký odhad v průběhu letu beze změny AIRSPEED_TRIM, tak ji aspoň nastavit na optimálnější hodnotu pro příští let. (Podobně jako se zachází s parametrem ASPD_SCALE). 

Obávám se ale, že tento problém nelze vyřešit bez znalosti i vertikální složky větru, která může kazit měření opadání, jak je již naznačeno výše.

kaklik, 2022-05-14:
Možná by tohle celé měření šlo nahradit automatickým odhadem některých parametrů TECS regulátoru, jako například FW_T_RLL2THR, nebo zavedením parametru Throttle_to_STE_rate tj. převrácenou hodnotu od [STE_rate_to_throttle](https://github.com/PX4/PX4-Autopilot/blob/3fe4c6e3f56d68417918454188a3f968c1e5ccd6/src/lib/tecs/TECS.cpp#L289), která se nyní určuje z konstant FW_T_SINK_MAX a FW_T_CLMB_MAX. Místo toho by zřejmě měla být vypočtena z dostupných výkonů během letu. 
Takové údaje by asi umožňily porovnání úprav virníků mezi s sebou ve smyslu horší/lepší bez toho aby bylo nutné měřit celou poláru.