ThunderFly-aerospace/TF-G2 #45: Povnost uchycení směrovky

kaklik, 2020-06-14:
zadní směrovka je teď citlivá na vylomení. Protože drží na dvou packách, které mají malý průřez a jsou spojeny čistě po vrstvách. 

Nepříjemné je zejména to, že směrovka v tomto místě vydrží podstatně méně, než zbytek dílu směrovky. Důsledkem pak je, že se ulomí packy spodního dílu, který se na ocasu mění v podstatě nejhůře. Navíc to velmi zkomplikuje výměnu, neboť snadno dojde k zalomení uhlíkové osy směrovky ve statickém dílu ocasu.

Správné vyvážení pevností by asi mělo vypadat tak, že nejdříve  (třeba rotorem) praskne náběžná hrana směrovky a tato prasklina se bude dál šířit, než směrovka úplně upadne. 
Neměly by ale upadnout packy nejhůře měnitelného dílu. 

Řešením by mohlo být tyto packy udělat zkosené v nějakém lichoběžníkovém tvaru, aby zůstaly tisknutelné, ale přitom byla zvýšena styčná plocha vrstev, protože asi není možné středový díl ocasu tisknout otočený kolmo oproti současné pozici. (aby vrstvy byly ve směru přenosu síly)

kaklik, 2020-06-14:
Společně s touto úpravou by zřejmě bylo vhodné vyřešit převážně estetický problém se zubem vznikajícím v napojení směrovky na horizontální část ocasních ploch.
Tato část by zřejmě mohla být vyřešena stejnou technikou, jako je řešeno napojení profilu frézovaných rotorových listů na plochý kořen listu.

roman-dvorak, 2020-06-18:
> Společně s touto úpravou by zřejmě bylo vhodné vyřešit převážně estetický problém se zubem vznikajícím v napojení směrovky na horizontální část ocasních ploch.
> Tato část by zřejmě mohla být vyřešena [...]

Tenhle problém s novou verzí středního dílu už není. Směrovka, co byla namontovaná na TF-G2+ už tímto problémem netrpí.

roman-dvorak, 2020-06-19:
Vyztužení těch pacek pro připevňení směrovky bylo provedeno tak, že v packách byla domodelovaná svislá díra. Ta má zajistit svislé perimetry přes problematickou vrstvu. Tedy celkové zpevnění a větší odolnost proti vylomení. 

Zároveň z dílu byly odbrány časti pro přišroubování serva. Servo bude v dílu zalepeno. To z důvodu, že zatím v ocasu používáme pokaždé jiné servo, které mají trochu jiné roměry. Materiál pro připevňení serva vytvářel nemalou část hmotnosti dílu. Zároveň byl upraven otvor pro protažení přívodních kabelů k servu. Nemá to zvláštní otvor ale je to integrované do kostky k servu. Ze stejného důvodu.

closed 2020-06-19 by roman-dvorak