ThunderFly-aerospace/TF-G2 #46: Bandáž pro akumulátory

kaklik, 2020-06-14:
Už opakovaně se stalo, že akumulátor byl poškozen o šroub na motorové přepážce. Je tak zřejmé, že toto místo bude vyžadovat konstrukční úpravu. Možnosti jsou:

- Zapustit šrouby do motorové přepážky
- na akumulátory vytvořit ohrádku, která bude akumulátor chránit před kontaktem se šroubem. 

Osobně bych asi preferoval nějakou formu zapuštění šroubů, nebo vytvoření krytu na tyto šrouby.  
Ohrákda na akumulátor totiž způsobí komplikace, při změnách akumulátoru za různé typy.

roman-dvorak, 2020-06-18:
Zapuštění šroubu je v tomto dílu problematické, motor za tyto šrouby tahá. Takže aktuálně tahá přes celý průřez dílu přes vrstvy, což je optimální řešení.

Navrhoval bych vytvořit nějakou krytku, která by se přišroubovala pod hlavy šroubů. a vytvářela by právě to zapuštění.  Zároveň by to mohla být krytka koukající osy od motoru.

kaklik, 2020-06-18:
A není potřeba se té přečnívající osy z motoru spíš zbavit?  Protože takhle překáží i akumulátorům.

roman-dvorak, 2020-06-18:
Určitě by to bylo lepší :)

kaklik, 2020-07-21:
Zatím vyřešeno zapuštěním šroubů.

closed 2020-07-21 by kaklik