ThunderFly-aerospace/TF-G2 #58: Roztáčení rotoru namotaným provázkem

kaklik, 2020-07-15:
Rotorovou hlavu by bylo potřeba upravit tak, aby bylo možné rotor manuálně roztočit na vyšší otáčky. 
K tomu by mohla posloužit změna, kdy by se rotor roztáčel provázkem namotaným okolo špulky na rotorové hlavě. 

Jako  základ špulky lze použít současný kalíšek určený k měření otáček rotoru.  Ten však v současné konstrukci neumožňuje přenášet kroutivý moment na rotor. 

Řešením zřejmě je kalíšek sjednotit se [spodním dílem šroubovaným na placku rotoru](Ten se šestihranným osazením pro matici rotorového šroubu), který se s rotorem spojuje dvěma šrouby. Tyto šrouby by mohly být použity pro přenášení momentu na rotorové listy.

slimonslimon, 2020-07-28:
Testované řešení: 
https://github.com/ThunderFly-aerospace/TF-G2/blob/master/CAD/png/printplate_888_4008.png

díky zužování špulky se velmi snadno dosáhne překřížení špagátku a jeho snadnému namotání, vystužený konec provázku dle mého soudu není potřeba. Momentálně se do špulky vejde cca 1.6m bílého provázku na vázání balíků. 

V rámci úpravy dílu: https://github.com/ThunderFly-aerospace/TF-G2/blob/master/CAD/png/888_1012.png pro uchycení Ir brány dojde k maximalizaci špulky. Při roztáčení na autě na platformě je možné tahat dál, než rozpětí rukou, ale menší silou.

kaklik, 2020-08-03:
Je pravda, že v současném řešení lze snadno dosáhnout překřížení provázku, ale  je to postup který je poměrně nedefinovaný. 
Proto se stále domnívám  že by byl lepší postup s vyztuženým koncem provázku, jako je například [použit zde.](https://www.youtube.com/watch?v=J7-8nUQytks)

Díl se špulkou by myslím chtělo upravit tak, aby průměr špulky byl zvětšený alespoň tolik, aby neinterferoval s otvory pro šrouby určenými ke stažení rotoru.  Tím navíc dojde ke snížení potřebné boční síly, takže by nemělo být potřebné rotor z boku tolik podpírat při tahání za provázek.

Dále by myslím značně pomohlo, pokud by před startem bylo možné namotaný provázek pojistit v nějakém zářezu, aby nemohlo dojít k jeho svévolnému rozmotání. Zde by se asi hodily dva symetrické zářezy na obvodu, jednak pak bude možné vybrat jeden z dvojice a potom touto úpravou nedojde k rozvážení rotorového domku. 

Při konstrukci tohoto roztáčecího mechanismu, by myslím bylo potřeba počítat s délkou provázku, která je znatelně menší, než rozteč rozpažených rukou většiny populace. Protože je nutné tento roztáčecí mechanismus používat i v případě startu z ruky.   Prakticky tak jde o délku provázku maximálně 1m. 

Celkově jsou potřeba tyto kroky: 

- [ ] Zvětšení průměru špulky pod rotorem na průměr větší než je rozteč šroubů rotoru
- [ ] Vytvoření otvoru pro zastrčení zpevněného konce provázku
- [ ] Vytvoření háčku do kterého lze namotaný provázek zatáhnout, aby nedošlo k jeho rozmotání před startem

kaklik, 2020-10-26:
Navrhoval bych roztáčecí bubínek upravit do této podoby: 

[image]

[Snímač otáček](https://github.com/ThunderFly-aerospace/TF-G2/issues/64) by pak nebyl z boku, ale ze spodní strany. 
Na bubínku by byl závit, který by zajišťoval jednak to že provázek nejde na bubínek namotat v protisměru a pak by měl také zabránit vzájemnému zapletení závitů. 

Předpokládám, že při roztáčení by se provázek z bubínku odvíjel ve smeru od horního okraje ke spodnímu.  Důvod k tomu je ten, že při nejvyšších otáčkách rotoru pak bude provázek nejvíce vzdálen od šroubů na kterých drží rotorové listy.

kaklik, 2020-12-25:
Model by ve spodní části zřejmě bylo možné odlehčit tak, že okolo matice, by byl vysoký kužel, jinde by materiál byl odebraný. 

[image]

Taktéž je zřejmě možné matici zapustit až skrz tento díl. Odpadl by pak problém s centrováním otvoru při skládání rotoru.

kaklik, 2021-01-01:
Z posledních pokusů vyplývá, že by bylo vhodné, aby pylon, nebo rotorová hlava ve svojí vrchní části obsahovala "hrazdu" přes kterou se přetáhne provázek namotaný na bubínek rotoru. 
Důsledkem by mělo být, že k roztočení rotoru by stačil tah za provázek především směrem dolů. Nebyl by tak nutný druhý člověk, který nyní tahá za provázek a drží rotorový pylon, aby jej nevyvrátil do boku. 

Hrazda by mohla být tvořena i bezzávitovou hladkou částí šnoubu. Ve složitější implementaci by byla lepší kladka s kuličkovým ložiskem. Výhodou použití ložiska s drážkovanou kladkou by bylo, že vydrží i mírný nesouosý tah bez snížení účinnosti.

slimonslimon, 2021-01-01:
případně by to mohla být celá soustava kladek k tažnému motoru, který by měl zařízení na tahání provázku v podobě dvou bubínků stlačených k sobě, takže by start vypadal tak, že by se přidáním plynu v prví fázy roztočil rotor předmotaným provázkem a pak už by se jen vírník vypustil z ruky

kaklik, 2021-01-01:
To s těmi dvěma bubínky nechápu, ale nápad je to zajímavý.  Akorát na tom tahání za provázek rukou se mi líbí že provázek pak zůstane na zemi a vírník ho netahá s sebou.

slimonslimon, 2021-01-01:
něco jako extruder v tiskárně, akorát místo filametu provázek a kapánek vyšší otáčky

kaklik, 2021-01-02:
> něco jako pohon filamentu v tiskárně, akorát kapánek vyšší otáčky

Aha! už to zřejmě chápu. Tím chceš vyřešit to, aby se provázek nenamotával na nějakou špulku u motoru, ale aby propadl na zem. No takovéhle řešení mě nenapadlo. :) Je to ale zajímavé protože takové řešení by mohlo vyřešit potíž s tvrdým rozběhem na začátku tím, že provázek mezi kolečky proklouzne.

slimonslimon, 2021-01-02:
Kdyby se tím chtěl někdo zabývat, tak je to takový hezký ucelený úkol - vzít rotor s měřákem otáček a tažný motor a zkusit někde "na stole" (v bunkru) roztahovat rotor tažným motorem... Já na to zrovna čas nemám :-(

kaklik, 2021-01-02:
> Kdyby se tím chtěl někdo zabývat, tak je to takový hezký ucelený úkol - vzít rotor s měřákem otáček a tažný motor a zkusit někde "na stole" (v bunkru) roztahovat rotor tažným motorem... Já na to zrovna čas nemám :-(

První krok podle mě stále je,  vyladit vůbec to roztáčení rotoru provázkem, tak aby to bylo bezchybný.  Pak se dá uvažovat o nějakém pohonu motorem. U motoru ale vidím výhodu, že ten provázek zřejmě může být i celkem dlouhý. :)

kaklik, 2021-01-18:
Aktuální provedení rotorové hlavy s provázkovou prerotací vypadá takhle: 

[image]

Provedení by mělo být kompatibilní s aktuálně používanou konstrukcí rotorové hlavy.

kaklik, 2021-02-08:
Při tisku jsem zjistil, že: 

- [ ] Výšku bubínku je potřeba o cca 1mm zkrátit, aby nedocházelo k drhnutí o spodní část rotorové hlavy
- [ ]  Horní šestihran pro plastovou matici šroubu udělat průchozí až ven, aby nebyla zbytečně využitá tak velká délka šroubu.

kaklik, 2021-05-03:
Řešení bylo otestováno na aerodynamické váze a zřejmě je funkční.

closed 2021-05-03 by kaklik