ThunderFly-aerospace/TF-G2 #59: Zvětšení rozlišení RPM senzoru

kaklik, 2020-07-15:
[Aktuální kalíšek](https://github.com/ThunderFly-aerospace/TF-G2/blob/master/CAD/stl/printplate_888_1014.stl), který slouží k měření otáček rotoru, má pouze 4 díry. 
Takový počet děr je dostatečný pro měření otáček jednou za sekundu.  Z experimentů ale vyplývá, že je potřeba otáčky snímat minimálně 4x za sekundu. Z čehož by při zachování rozlišení vyplýval počet děr alespoň 16.

Úprava kalíšku na takto zvetšený počet děr je reálná i při zachování současného průměru.  Myslím si však, že průměr kalíšku je potřeba trochu zvětšit. 
Je to důležité ze dvou důvodů. Jednak proto aby se do něj snáze vešla optická závora. Druhý důvod je, že kalíšek je pravděpodobně potřeba upravit tak, aby [umožňoval roztočení rotoru kusem špagátu](https://github.com/ThunderFly-aerospace/TF-G2/issues/58).

roman-dvorak, 2020-07-15:
Pro použití pro roztáčení rotoru (#58) by to chtělo celkově změnit konstrukci tohoto dílu. Problém tohoto dílu je, že oproti ose rotoru drží za velmi malou plochu a při roztáčení za provázek by se to začalo velmi snadno protáčet. Řešení tohoto problému je několik. 

Aktuální verze má opravené dření o optický senzor a zvětšen počet děr na 16.

kaklik, 2020-07-15:
@slimonslimon  To vypadá, že tohle issue související s #58 je vyřešeno..

kaklik, 2020-07-28:
Řešení bylo otestováno společně s úpravou pro roztáčení rotoru  a je funkční pro otáčky rotoru do ~2000 RPM,

closed 2020-07-28 by kaklik