ThunderFly-aerospace/TF-G2 #60: Úprava rotorových podložek

kaklik, 2020-07-16:
Rotorové podložky je potřeba upravit tak, aby seděly k nové rotorové placce #57.

cesilko1, 2020-07-16:
[image]

kaklik, 2020-08-18:
Momentálně je tohle issue vyřešeno dočasně tím že podložky jsou stále tvořeny polygonem, který ale v důsledku zřejmě není parametrický.  

To znamená, že například nelze: 

- Nastavit mezeru v podložkách pro konkrétní úhel prověšení listů před startem
- Rozteče šroubů a pozice jsou pravděpodobně momentálně dány číselnými hodnotami v kódu, těžko se tak díly provazují vzájemně, nebo vytváří závislosti na dalších rozměrech.

kaklik, 2021-08-04:
Tohle bych zatím navrhoval neřešit a počkat jak dopadne free-flapping-rotor od @slimonslimon

kaklik, 2021-08-21:
Výsledkem úspěšného testu free-flapping rotoru je, že není potřeba tento úkol dále řešit.

closed 2021-08-21 by kaklik