ThunderFly-aerospace/TF-G2 #62: Optimalizace roztáčení rotoru při startu

kaklik, 2020-08-17:
Při startu vírníku se dlouhodobě potýkáme s problémem nalezení optimální rychlosti jízdy vůči aktuálním otáčkám rotoru a záklonu rotoru vůči proudu vzduchu. 

K vyřešení jsou potřeba dva kroky: 

1. Nalézt metodiku, jakým způsobem volit záklon rotoru a rychlost jízdy v závislosti na aktuálních otáčkách, aby docházelo k optimálnímu zvyšování otáček, bez zbytečného namáhání rotorové hlavy a listů. 
2. Rozsah optimální rychlosti zobrazit řidiči (Zde například nějakou kruhovou výsečí v ukazateli airspeed).

kaklik, 2020-08-17:
K té metodice startu, si to já představuji tak, že by záklon rotoru určoval pilot pomocí ovládání pitch-setpointu (jako to dělá do teď). Hlavní rozdíl by byl v tom, že výstup pitch z autopilota, abychom potřebovali mít v aplikaci na zemi. (Aby se z toho dal přepočítat aktuální úhel rotoru vůči proudění). 

Pak následně pozemní aplikace by měla zobrazovat optimální rychlost pro řidiče.  Zde předpokládám, že z nejnižších otáček rotoru je dobré se dostávat maximálním záklonem rotoru a pomalou jízdou.  Záklon zde zajistí pilot. 
Potíž ale je, že pilot potřebuje, aby při odpoutání vírník letěl rovně ve vyrovnané poloze. Což znamená, že musí rotor během startu předklonit dopředu a řidič musí zároveň adekvátně zvýšit rychlost jízdy. A tyto dva procesy je nutné koordinovat pomocí ukazatele rychlosti pro řidiče.

kaklik, 2020-08-18:
Tak mě napadlo, že možná lze pro první přiblížení tu závislost na záklonu rotoru vypustit a pouze zobrazovat maximální rychlost vzhledem k otáčkám rotoru. 

Tenhle ukazatel by pak jen škáloval AIRSPEED_TRIM vůči aktuální/nominální otáčky rotoru. 

Tj, na nominálních otáčkách rotoru musí být dosaženo AIRSPEED_TRIM.

roman-dvorak, 2020-09-18:
Rovnice přepsaná z předchozích obrázků

![rovnice](https://render.githubusercontent.com/render/math?math=v=(\sqrt{\frac{2}{3}%20\frac{m}{\rho*T*C_L}%20-%20\frac{l^2}{18}}%20-%20\frac{2}{3}l)%20*%20\omega)

kaklik, 2020-10-12:
Tak sme ověřili, že tohle řešení je zřejmě správná cesta a rotor se tak roztáčí poměrně efektivně. Přičemž při dodržení rychlostí nebylo pozorováno extenzivní namáhání rotoru.

Potíž však je, že při rozjezdu ke startu dochází k přetáčení rotoru. Což je způsobeno tím, že současná implementace ukazatele nereflektuje omezení na rychlost jízdy popsané v [předchozím komentáři](https://github.com/ThunderFly-aerospace/TF-G2/issues/62#issuecomment-675690180). 

Řidič tak při sledování ukazatele snadno překročí vhodnou maximální rychlost, čímž se nadále zvyšují otáčky rotoru.  Řešením je ukazatel upravit tak, aby zelená kruhová výseč ukazující optimální rychlost jízdy nemohla překročit maximální letovou rychlost UAV. Konkrétněji by se měla zastavit na optimální startovní rychlosti.