ThunderFly-aerospace/TF-G2 #64: Aktualizace snímače otáček rotoru

kaklik, 2020-09-21:
[Aktuální snímač otáček](https://github.com/ThunderFly-aerospace/TFRPM01) je použit v kombinaci s optickou závorou. 
Na TFRPM01B je přidán pull-up ?3k? a fototranzistor je připojený na stejný vstup jako se připojuje hallova sonda. 

Použitá optická závora je náhodného typu ze zásob Robozoru. Je připevněna na vertikální držák ložisek rotorové osy. Skrz závoru se otáčí kalíšek s děrovanými otvory.

kaklik, 2020-10-26:
V případě že by pro snímání otáček, byl použit reflexivní snímač, tak existují dva způsoby jeho použití. 

### Snímač umístěný rovnoběžně se směrem otáčení

V této pozici je vysílač a přijímač reflexivního snímače umístěn podélně se směrem rotace.  Což odpovídá následujícímu obrázku. Pohled na rotorovou hlavu ze zadní strany, kde plošný spoj se snímačem je uchycen mezi rameny táhel rotorové hlavy. 

[image]

### Snímač umístěný radiálně

V tomto umístění by vysílač a přijímač reflexního snímače byly orientovány v radiálním směru vůči ose rotace rotoru. 
Umístění je ilustrováno na následujícím obrázku, který reprezentuje pohled z boku na rotorovou hlavu.

[image]

V obou případech je reflexivita bubínku zajištěna spodní plochou bubínku, která při tisku bude položena na tiskové podložce. 
V této ploše jsou tmavé otvory realizované dutinou, která může přecházet v celý objem bubínku. Na obrázcích výše jsou znázorněny příčky, které mají sloužit ke zpevnění bubínku vůči ose rotace.

kaklik, 2020-10-26:
@roman-dvorak  
Tohle vypadá na to, že repozitář vůbec neobsahuje reálně použitou verzi měřícího děrovaného bubínku rotoru. 

[image]

kaklik, 2020-11-21:
Zatím jsem vytvořil přípravu pro uchycení TFPROBE.

[image]

roman-dvorak, 2020-11-24:
@kaklik ... Jak se sem strká ta matka? :D

[image]

Navrhoval bych vytvořit kapsu.

roman-dvorak, 2020-11-24:
Při doražení PCB do konce, drážky PCB naráží na šroub. Děje se tak i v případě, že pod hlavou šroubu není podložka. 

Bylo by dobré, aby v drážce byla zarážka, která by definovala maximální zastrčení PCB bez toho, aby to naráželo do šroubu roll osy rotorové hlavy.

kaklik, 2020-11-25:
Já úplně nemám nápad, jak zajistit, aby šlo rozmontovat Roll osu a přitom tam mohl zůstat ten PCB.  Předpokládal jsem že se ten PCB bude muset vyndavat. Protože při montáži tam dost vadí hřebínek který trčí dolů, jednoduché posunutí PCB výše proto moc nepomůže.

Co se týká řešení té lámací hlavy, tak jsem přemýšlel o rozdělení tohoto dilu na dva díly, které by se šroubovaly k sobě.  Tím by se po odlomení teoretecky nemusela odšroubovávat táhla k servům. 

S tím vkládáním matice šroubu je to blbý, ale nejsem si jistej, zda je možné v kombinaci s kapsou používat samojistnou matici.

roman-dvorak, 2020-11-25:
> Já úplně nemám nápad, jak zajistit, aby šlo rozmontovat Roll osu a přitom tam mohl zůstat ten PCB. Předpokládal jsem že se ten PCB bude muset vyndavat. Protože při montáži tam dost vadí hřebínek který trčí dolů, jednoduché posunutí PCB výše proto moc nepomůže.

OK, neřešme to. Vyndat PCB není složité :) 

> Co se týká řešení té lámací hlavy, tak jsem přemýšlel o rozdělení tohoto dilu na dva díly, které by se šroubovaly k sobě. Tím by se po odlomení teoretecky nemusela odšroubovávat táhla k servům.
Jo, tohle je fajn nápad! Mělo by to být součástí #69 

> S tím vkládáním matice šroubu je to blbý, ale nejsem si jistej, zda je možné v kombinaci s kapsou používat samojistnou matici.
Navrhoval bych to zkusit. Protože jsem dneska zkoušel tu matku tam vložit a je to opravdu složité. 

>Při doražení PCB do konce drážky, PCB naráží na šroub. Děje se tak i v případě, že pod hlavou šroubu není podložka.
Tento problém nebude nastávat, protože PCB stejně nebude tak hluboko v dílu.

kaklik, 2020-12-24:
Upravený bubínek s vnitřními zuby pro snímání otáček rotoru [TFRPM01B](https://github.com/ThunderFly-aerospace/TFRPM01) v kmbinaci s [TFPROBE](https://github.com/ThunderFly-aerospace/TFPROBE01).

https://user-images.githubusercontent.com/1194177/103105607-1edae000-462f-11eb-8e46-ee450d7dc4fb.mp4

kaklik, 2020-12-25:
Ta zarážka na PCB TFPROBE momentálně postrádá význam, protože PCB musí být zaražený v drážce až k hlavě šroubu, aby to fungovalo správně..

kaklik, 2021-01-17:
Poslední iterace vypadá takto:

[image]

Myslím, že řešení je již použitelné.  Další zlepšení je možné až po dalších úpravách jako je #69. 

Řešení je nyní kompatibilní s původní rotorovou hlavou TF-G2  @slimonslimon.

kaklik, 2021-02-19:
Čidlo po posledních úpravách funguje i těsně vedle 60W žárovky. Test na přímém slunečním světle je potřeba ještě provést, ale domnívám se, že problém je vyřešen.

kaklik, 2021-03-27:
Na základě poznatků z issue: https://github.com/ThunderFly-aerospace/TFRPM01/issues/10 to vypadá, že je potřeba zvýšit barevný kontrast měřícího kloboučku. 

Jedna z možností, jak to vyřešit je vzít nějaký "IR tmavý" materiál. Například modrou, nebo šedou a na kloboučku vytištěném z této barvy obarvit pouze měřící zuby bílou fixou.

kaklik, 2021-03-28:
[image]

roman-dvorak, 2021-03-28:
Skvělé.. tohle vypadá, že půjde realizovat relativně snadno.

kaklik, 2021-03-28:
U modrého kloboučku je situace o něco horší, kvůli nižší hodnotě maximální hodnoty napětí.  To je na hranici funkčnosti, těsně pod hodnotou 2V. Tuto situaci by bylo možné zlepšit snížením hodnoty pull-upů buď na TFPROBE, nebo na TFRPM. 

[image]

Nicméně i při současném nastavení, snímač ještě stále měří.  Jak je z grafu vidět, tak vysoká hodnota má trochu rozptyl, závislý na přesné poloze kloboučku. To je zřejmě způsobeno tím, že je měření provedeno na starší verzi kloboučku, která ještě nemá tak přesně umístěné zuby.  Je pravděpodobné, že s novou verzí by byl signál lepší.

kaklik, 2021-04-08:
Vyzkoušel jsem ještě novou tvarovou úpravu kloboučku, která má eliminovat výše popsanou nedokonalost způsobenou vyosením zubů. V podstatě to zřejmě zlepšuje, ale ne výrazně. 

[image]

Způsob snímání ale pokládám za funkční.

closed 2021-04-08 by kaklik