ThunderFly-aerospace/TF-G2 #65: Pohonné ústojí vírníku

roman-dvorak, 2020-09-21:
Pro opakovatelnost výroby vírníků TF-G2 (TF-G3) je potřeba vybrat spolehlivé a dostupné ESC regulátory, motory a vrtule. 

# ESC 
Na regulátory jsou požadavky
 * Telemtrická data přes UAVCAN
 * Dostupnost
 * Výkon ~50A (@ 3S) 
 * OpenSource (výhoda, pokud ne tak reálná měření)
 * 

# Regulátor

## [Myxa](https://zubax.com/products/myxa)
Regulátor Myxa od Zubaxu. 

Klady
* Odpovídající výkon
* UAVCAN (plná podpora včetně nastavování přes CAN) 
* Malý rozměr, malá hmotnost
* Zubax vypadá jako docela komunikující firma, aktivní vývoj
* OpenSource, několik variant
* Podpora PPM + UAVCAN ovládání

Zápory
* Vyžaduje 4S akmumulátor
* Nemá BEC

# Motor
Požadavky:
 * 

# Vrtule
Požadavky:
 * Známé parametry
 * Dostupnost
 * Cena (spotřební materiál)

Jako součást s nejmenší možností výběru je je asi regulátor. Při volbě Myxi jsme omezeni především na to, že nejnižší podporované napětí splňuje 4S akmululátor. Od toho se odvíjí výběr zbylých komponent.

roman-dvorak, 2021-06-26:
Po několika pokusech se ukazuje, že MYXA pro TF-G2 (4S) dostačuje ale s nevelkou výkonostni rezervou.

Stále tedy v úvahu připadá používání VESC regulátoru. Na motorové váze bude vhodné tyto dva regulátory porovnat a případně se pokusit myxu nějakým způsobem lépe nastavit.

kaklik, 2021-07-06:
Nevím sice na jakém konektoru je u [nového VESC](https://www.neven.cz/kategorie/elektronicke-soucastky/stmivace/pwm-regulator-otacek/regulator-vesc-bldc-superfoc6-8-50a-esc/) RC PWM vstup. Nicméně jsem zjistil, že zřejmě nemáme sedmipinový konektor.  Není tak možné jeden z konektorů na PCB použít. Konektory, ke kterým máme protikusy jsou vyfoceny na následujícím obrázku. 

[image]

kaklik, 2021-09-04:
Tohle issue je zřejmě potřeba rozebrat na dílčí části, které je nutné řešit samostatně.

closed 2021-09-04 by kaklik