ThunderFly-aerospace/TF-G2 #67: Krabička pro akumulátor

kaklik, 2020-09-24:
### Hmotnost krabičky

Krabička pro akumulátor je nyní zřejmě zbytečně masivní. Neboť akumulátor je zároveň částečně chráněn i podvozkem. K poškození akumulátoru by tak muselo dojít po tom, co bude zničen podvozek. 

Krabička pro akumulátor tak může být méně masivní a tím v důsledku i lehčí. 

### Uchycení krabičky

Krabičku nebo spíše kryt akumulátoru by bylo dobré na tělo vírníku upevňovat tak, aby nebylo nutné používat nástroje. 
Jedno z řešení které mě napadá je na krabičku vytvořit západky, které se zasunou do protikusů na těle vírníku.  Krabička by se pak zajistila proti vypadnutí zasunutím napájecího konektoru/konektorů do panelového protikusu na těle vírníku.

roman-dvorak, 2020-11-24:
Usnadnění montáže/demontáže bylo vylepšeno použitím šroubů s velkou hlavou, které lze utáhnout/povolit rukou. 

Jedná se o šroub M3 DIN 464

roman-dvorak, 2020-11-25:
Odlehčení proběhlo ve větvi 4S. Je to parametrické, takže to funguje i s menší verzí krabičky. Zároveň je to příprava na zamontování přírubového konektoru od ESC do trupu vírníku pro snadné připojení/odpojení.

closed 2020-11-25 by roman-dvorak