ThunderFly-aerospace/TF-G2 #69: Lámací rotorová hlava

kaklik, 2020-10-31:
V některých případech se stává, že se z rotorové hlavy oddělí pouze jeden list. Buď je ho předchozím poškozením, nebo nárazem při havárii. 
Důsledkem pak je utržení celého pylonu druhým nepoškozeným listem.  Trhací podložky tomuto stavu nedovedou zabránit, neboť jsou konstruovány tak, aby v tahu přenesly relativně velkou sílu a ke svému zlomení potřebují moment protilehlého listu. 

Řešením by mohla být úprava ložiskového domku rotorové hlavy tak, aby došlo při extenzivní boční síle k odlomení celého šroubu rotoru i s ložisky. V běžném režimu je zde boční síla téměř nulová, až na vibrace a anomální letové stavy. Při odlomení jednoho z listů boční síla radikálně vzroste.

roman-dvorak, 2020-11-25:
V #64 @kaklik navrhl následující řešení:

> Co se týká řešení té lámací hlavy, tak jsem přemýšlel o rozdělení tohoto dilu na dva díly, které by se šroubovaly k sobě. Tím by se po odlomení teoretecky nemusela odšroubovávat táhla k servům.

roman-dvorak, 2021-05-14:
@Alsa-Tucnak Je nějaký pokrok s touto věcí? 
v jaké větvi je to plánované namodelovat?

kaklik, 2021-05-15:
[image]

Takhle od oka je ta drážka přibližně dobře, měla by být 4mm pod přepážkou.  Protože ložisko je horní stranou doraženo na zub, který vytváří přepážka (ta se po tisku odvrtá) a samotné ložisko [623 SKF](https://www.skf.com/group/products/rolling-bearings/ball-bearings/deep-groove-ball-bearings/productid-623-2Z) je  4mm vysoké. 

Podstatným zlepšením by ale bylo, kdyby ulamovací díl byl vyměnitelný a šrouboval se k základně s táhly k servům.

kaklik, 2021-06-16:
@Alsa-Tucnak  nyní rozdělila rotorovou hlavu na dva díly 

[image]

[image]

Myslím že řešení je docela dobré, akorát šestihran na matice šroubů pro uchycení sloupku s ložisky musí být na spodní straně základny. Zde:

[image]

roman-dvorak, 2021-06-25:
Po pokusu o sestavení kompletní rotorové hlavy s novým trhacím ložiskovým domkem jsme přišli na tyto problémy: 
Sloupky pro mlýnek byly uříznuté.

kostka, ve které jsou svislé šrouby pro odmontování trhací části, zasahuje do rolničky, kterou díky tomu nelze hladce otáčet.  
[image]

Celý díl je nyní vyšší. Což se projevuje mezerou pod rolničkou. 
[image]

[image]

Napadají mne dvě řešení:
 * Snížit sloupek, ve kterém jsou ložiska, do původního stavu. Ideálně tak, aby se zachovala původní rozteč ložisek. Pokud se změní vzdálenost ložisek, tak je potřeba vybrat vhodný koupitelný pevnostní M3 šroub a jejich vzdálenost musí být taková, aby mezi ně existoval vhodný rozpěrný sloupek (nebo ho tisknout).  
 * Pokud je nějaký vážný důvod zachovat tento zvětšený sloupek, tak je potřeba zachovat původní geometrii rotorové hlavy viz. obrázek. Původní rotorová hlava to splňuje. 
[image]

Z důvodu těchto problémů s touto hlavou nebyl proveden žadný letový pokus.

kaklik, 2021-06-26:
Ještě sem doplním, že test zlomení vyměnitelného dílu dopadl dobře v tom smyslu, že se zlomil na místě, kde je zápich a síla potřebná ke zlomení mi přijde přibližně adekvátní zamýšlenému užití.

[image]

Alsa-Tucnak, 2021-07-22:
> 
> 
> Po pokusu o sestavení kompletní rotorové hlavy s novým trhacím ložiskovým domkem jsme přišli na tyto problémy:
> Sloupky pro mlýnek byly uříznuté.
> 
>     * [ ]  kostka, ve které jsou svislé šrouby pro odmontování trhací části, zasahuje do rolničky, kterou díky tomu nelze hladce otáčet.
>       [image]
> 
>     * [ ]  Celý díl je nyní vyšší. Což se projevuje mezerou pod rolničkou.
>       [image]
> 
> 
> [image]
> 
> Napadají mne dvě řešení:
> 
>     * Snížit sloupek, ve kterém jsou ložiska, do původního stavu. Ideálně tak, aby se zachovala původní rozteč ložisek.
> 
>     * Pokud je nějaký vážný důvod zachovat tento zvětšený sloupek, tak je potřeba zachovat původní geometrii rotorové hlavy viz  následující obrázek. Původní rotorová hlava to splňuje.
> 
> 
> [image]
> 
> Z důvodu těchto problémů s touto hlavou nebyl proveden žadný letový pokus.

Nevím, jak došlo ke zvýšení hlavního válce. Pokud je to opravdu vyšší, tak je to omylem.

roman-dvorak, 2021-07-23:
Výška nyní vypadá správná. Bohužel jsem včera zapomněl si odvézt nějakou složenou rotorovou hlavu, takže to neotestuji kompletně.

Ale má to problém, že šrouby, které stahují ty dva díly k sobě, kolidují s kódovacím diskem na rolničce. Nepůjde tak s rolničkou (rotorem) volně otáčet. 

Řešením by bylo posunout šrouby do osy rotoru. 

[image]

[image]

Po vytištění má sloupek ze spodní strany nějak divně vytištěné mosty. Zřejmě je to způsobené kulatou dírou v horní straně. Bylo nutné to očistit. Jinak byl sloupek šikmo.

[image]

kaklik, 2021-08-01:
Aktuální řešení bylo ověřeno praktickým letem, jako funkční. :)

closed 2021-08-01 by kaklik