ThunderFly-aerospace/TF-G2 #70: Otočení osy serv

kaklik, 2020-10-31:
Osu serv na pylonu je potřeba otočit po směru letu.

roman-dvorak, 2020-11-24:
Aktuální řešení vypadá takhle. K vyřešení tohoto isuue navrhuji 2 úpravy. 

 * Rozšíření (hloubky) sloupku pro připevňení serva. 
 * Nepatrně zvetšit průměr otvoru pro samořezný šroub. 

[image]
[image]

roman-dvorak, 2020-11-24:
[image]

closed 2020-11-24 by roman-dvorak

kaklik, 2020-11-25:
Trochu jsem nečekal, že ta serva budou z vnitří strany pylonu..   Protože takhle ta táhla jsou pod docela odlišným úhlem než kolmo na osu pohybu.  Ale fungovat to asi bude.

roman-dvorak, 2020-11-25:
Ony se na vnitřní stranu tisknou kapsy na matky. Z druhé strany se to takto  udělá těžko. 

Spíše je blbé, že ty páky dodávané se servy jsou na obě strany. Což znamená, ze se musí uříznout a po jejím připadném zlomení už není možnost tu páku pouze otočit.

kaklik, 2020-11-25:
Kdyby ta serva, ale byla z venku, tak tenhle problém s pákami odpadá.  Kapsy na matky jsou myslím vytisknout i směrem k podložce. 

Případně jde myslím prohodit ten šedý a modrý díl tak, že kasy by byly v tom šedivém horizontálním dílu.

roman-dvorak, 2020-11-25:
ten šedý díl tam tvoří věc, že horní a spodní čast pylonu mají trochu jiný úhel vůči sobě. Ten rámeček je taková přechodka mezi těmito úhly. 

Tahle skladba tehdy měla nějaké opodstatnění. Teď už nevím jaké přesně.

roman-dvorak, 2020-11-25:
V tom případě by se asi chtělo zamyslet nad nějakým komplexnějšim předěláním pylonu.