ThunderFly-aerospace/TF-G2 #75: Zmenšení pevnosti na lom rotorových trhacích placek

roman-dvorak, 2020-11-24:
Aktuální verze rotorových placek je velmi pevná na lom. Díky tomu se ničí velké množství listů.

@kaklik navrhl, aby zářez byl trojúhelníkový. Aby na vnítřní straně byl ostrý úhel. 

Trhací placky mají být první deformační zóna. Druhou má být trhací rotorová hlava: #69

roman-dvorak, 2020-11-24:
S tímto issue je efektivní vyřešit i #74 u dílu trhací placky

kaklik, 2020-11-25:
Tak jsem otočil smysl středových děr, čímž se popisky myslím dostaly na vrchní stranu ve složeném rotoru. 

[image]

A taktéž trhací otvory vyměnil za 4hrany. Zde trochu váhám, jestli na rotoru neexistuje nějaká asymetrie sil při nárazu listu, která by se dala využít k lepšímu utržení listu.  Přijde mi ale, že když se jeden list dotkne země, tak síla která je vyvozená druhým stále rotujícím listem, na jeho závěs je symetrická.

roman-dvorak, 2020-11-26:
Zřejmě je zeslabení nedostatečné.

Nepomohlo by trhací otvory přesunout do flapovací spáry? Protože v tomto místě je placka stažená docela velkou silou, z obou stran, rotorovým listem a spodní podložkou.

kaklik, 2020-11-27:
Ve flapovací spáře se ale místo předpokládaného roztržení bude cyklicky namáhat. Takže zde bude postupně vznikat prasklina i bez namáhání listů silami, které mají vést k utržení. 

Navrhoval bych snížit počet perimetrů trhací podložky. Zdá se že sílu přenáší v místě roztržení hlavně perimetry.

kaklik, 2021-02-17:
Na základě posledních testů mi přijde, že toto issue je již irelevantní, neboť k utržení listů po dopadu docházelo i bez jejich znatelného poškození.

commit a59cde56 (2020-11-25): Fix #75
commit 619069d8 (2020-12-01): Fix #75

(cherry picked from commit a59cde56f261afa749246cdc747a7ac4d992294c)
closed 2021-03-09 by kaklik