ThunderFly-aerospace/TF-G2 #76: Zlepšit umístění RPM PCB

roman-dvorak, 2020-11-24:
Bylo by dobré, do poličky uprosřed pilonu vylepšit připevňení senzoru tak, aby matka v něm byla zapuštěná (tj. aby nebyl potřeba klíček a umístit ho více do středu tak, aby hrana PCB nekolidovala s bočnicemi pylonu. 

Zároveň by nebyla špatná nějaká krabička, která by to zakrytovala před vodou. Zároveň je potřeba vyřešit nějaké svedení všech kabelů z rotorové hlavy.

kaklik, 2020-11-25:
Asi bych potřeboval nějaký obrázek, abych pochopil, co tohle issue znamená.

roman-dvorak, 2020-12-02:
Původně to bylo myšleno tak, že se přesune senzor do prostoru střední příčky pylonu, kde se na něj vytvoří nějaký "držák". 

Nyní bych navrhoval TFRPM dát někam do spodní části vírníku (někam do prostoru ESC a autopilota). Pro začátek asi v TFRPM01 v krabičče, ve které je dodávaný. 

Tohle z důvodu, že RPM kablík se snadno vytrhne z rotorové hlavy (z TFPROBE) a nebude se tahat za I2C vedení při možné nehodě.

kaklik, 2021-02-17:
A jak to teď vlastně je s použitím TFRPM driveru na stejné sběrnici jako TFSLOT?  Funguje to, nebo ne? 

Protože k TFSLOT se stejně I2C na pylon bude tahat, takže nevidím moc rozdíl v tom tam vést dvakrát I2C, nebo I2C a servokablík. 

Z hlediska HW by bylo nejlepší řešení jen jedno I2C na které by byly připojeny oba senzory, díky tomu, že TFRPM je průchozí.  Vím ale že dříve jsme s tímto řešením měli potíže. Je tak otázka zda je to však relevantní i pro současný software?

slimonslimon, 2021-02-18:
Tady asi chybí nějaké "měření/pokusy" který by zjisti, kde je problém. 

Driver neprovádí novou inicializaci RPM senzoru. Ale je potřeba vyzkoušet že reset rpm driveru stačí na obnovení funkčnosti.

roman-dvorak, 2021-02-23:
Na vírníku 4S tohle bude TFRPM integroaný do blízkosti TFSLOT. Proto je toto issue již nerelevantní.

kaklik, 2021-03-22:
Issue lze zavřít, až když bude reálně vyřešené.

kaklik, 2021-04-02:
> Tady asi chybí nějaké "měření/pokusy" který by zjisti, kde je problém.
> 
> Driver neprovádí novou inicializaci RPM senzoru. Ale je potřeba vyzkoušet že reset rpm driveru stačí na obnovení funkčnosti.

Z [tohoto experimentu](https://github.com/ThunderFly-aerospace/TF-G2/issues/88#issuecomment-785433829) vyplynulo, že snímače na jedné sběrnici fungují a čtení nepadá. Co ale nefunguje je měření otáček, které při takovém zapojení občas dává zápornou hodnotu otáček..