ThunderFly-aerospace/TF-G2 #79: Odvibrování autopilota.

roman-dvorak, 2020-11-24:
Autopilot je za letu vystaven docela velkým vibracím:
[image]

Navrhoval bych autopilota umístit na takovéto silenbloky: 
https://www.tme.eu/cz/details/dva2-8-8-m3-40/tlumice-chveni/elesa-ganter/dva-2-8-8-m3-6-40/

https://www.tme.eu/cz/katalog/tlumice-chveni_112550/?visible_params=2%2C1482%2C205%2C142%2C2636%2C1209%2C532%2C1501%2C2779%2C2599%2C531%2C2892%2C329%2C138&mapped_params=532%3A1448207%3B

Oproti řešení z AutoG2 je zde výhoda nějaké normální montáže. Nebude hrozit opakované vypadávání atd.. Zaroveň to vypadá jako opakovaně kopitelná věc. Existují různé konfigurace šroubu/otvoru se závitem. Já bych navrhoval variantu na jedné straně otvor se závitem a na druhé straně šroub (tj ta, co je v prvním odkazu)

ChroustJan, 2020-11-25:
@roman-dvorak mám jich vzít 10ks: https://www.tme.eu/cz/details/dva2-8-8-m3-40/tlumice-chveni/elesa-ganter/dva-2-8-8-m3-6-40/  ?   Jsou potřeba akutně nebo stačí třeba za týden?

roman-dvorak, 2020-11-25:
@ChroustJan 
Nejsou potřeba akutně. Ale bylo by dobré je zařadit do dalšího nákupu. Rád bych je použil do vírníku se 4S akumulátorem, axi motory atd.. Takže ten týden, dva, je fajn termín.

kaklik, 2020-11-25:
Mě přijde že tuhost 22N/mm je strašně moc.. Že by to chtělo tak maximálně poloviční hodnotu.   Představte si, že na ten sloupek zavesíte 2L flašku vody a on se natáhne o milimetr.. 

Z toho výběru v TME by šlo možná použít [tohle 10N/mm](https://www.tme.eu/cz/details/dva1-10-15-m4-40/tlumice-chveni/elesa-ganter/dva-1-10-15-m4-10-40/).  Ale má to šroub M4, což je pomerně nekompatibilní s tím co teď používáme. Navíc má šrouby z obou stran.

Stejné kostrukční provedení jako navrhuje @roman-dvorak  má [tohle s 11N/mm](https://www.tme.eu/cz/details/dva2-10-15-m4-40/tlumice-chveni/elesa-ganter/dva-2-10-15-m4-10-40/). 

Myslím, že by to mohlo jít vyzkoušet i se šrouby M4, zda je to vůbec použitelné. 

Další možnost jsou M3 silenbloky tady: https://www.proprumysl.cz/hledani?q=silentblok+M3

kaklik, 2020-11-29:
Je zajímavé jakým způsobem jsou na airframe vírníku rozloženy vibrace.  Měřené osy jsou na vírníku orientovány takto: 

[image]

Amplituda vibrací ve zrychlení podél směrových os ale vypadá takto: 

[image]

V případě rotací podél směrových os naopak takto: 

[image]

Z naměřených hodnot je vidět, že předozadní vibrační směr se na autopilotovi uplatňuje poměrně minimálně. S autopilotem je nejvíce vibrováno do stran a vertikálně. 

V případě rotačních zrychlení je situace opačná. Autopilot se nejvíce otáčí okolo předozadní osy, následně podél vertikální osy a naopak je téměř stabilní okolo příčné osy. 

Z tohoto pozorování vyplývá že autopilot může být poměrně tuhým způsobem uchycen zpředu a zezadu. Musí mít tak možnost pohybu ooproti trupu vírníku do stran a vertikálně. Rotační uchycení musí mít nejmenší tuhost podél předozadní osy.

kaklik, 2020-12-14:
Možná že tohle issue je již irelevantní vzhledem k existenci požadavku na úpravy pylonu #82 .