ThunderFly-aerospace/TF-G2 #82: Upravený pilon

roman-dvorak, 2020-12-02:
- [ ] Spodní rámeček 888_1007
 - [ ] Spodní část držáku silentbloků 888_1008

 - [ ] Vybrat + otestovat správné silentbloky
 - [ ] Horní část držáku silentbloků
 - [ ] Implementace Airspeed #16 
 - [ ] Umístění serv #70 
 - [ ] Rotorová hlava
 - [ ] Trhací pylon

<hr> 

Protože doposud nedošlo k výrobě vírníku v podobě frézovaných (nebo laserem pálených) dílů, tak bych odstoupil od tohoto způsobu výroby. Trochu bych se inspiroval tyčovým pylonem na TF-G1. 

Uprostřed kostry vírníku by byl rámeček, který bude zpevňovat trup a bude pevným základem pylonu. V něm by byly 4 uhlíkové tyčky průměru 6 nebo 7 mm (jako jsou aktuálně na platformě). Ty povedou do druhého rámečku. V rámečcích budou zalepeny. Skládání bude probíhat tak, že se tyčky prostrčí ze spodní strany do horního rámečku. Tím bude skládání snazší než u pylonu TF-G1. 

Do horního rámečku budou zašroubovány 4 silentbloky, které budou fungovat jako odpružení/odvibrování pylonu. Ovládací prvky budou z důvodu tuhosti řízení v horní části rotorové hlavy. 

Horní část ještě nemám přesně promyšlenou. Ale mělo by to vypadat tak, že bude co nejnižší a nejkompaktnější. Stále se počítá s uspořádáním serv na boku. Protože to snižuje podélnou plochu pylonu a zároveň tuhost řízení. 

Také jsem přemýšlel o rotorové hlavě založené na kulovém čepu. Ale s tím si nejsem jistý. 

Další otázka je implementace trhacího pylonu. Nabízí se několik řešení. Nejvíce se mi líbí, že by se vytrhly silentbloky v nějakém definovaném - oslabeném místě. (I tak se rotor má prioritně  trhat v listech #75 popř. domku ložisek rotoru #69 :smile: ) 

Očekával bych, že by se těmito změnami zvětšila pevnost celé konstrukce.

kaklik, 2021-07-05:
Tak mám teď po montáži nových zařízení na střechu měřícího auta a zároveň z průběhu testů tomu několik domnívám se zásadních poznatků. 

1. Vypadá to, že aerodynamický odpor na střeše auta zvětšují zejména kabely, tedy tenké objekty v malém kruhovém průřezu zasahující do proudu vzduchu. Je to sice poměrně neexaktní měření, ale změna je po přidání kabelů myslím i slyšitelná. 
2. Zdá se že realizace vícetyčkového pylonu s tyčkami malého průřezu není dobře realizovatelná. Protože je problém s rozkladem sil do větší plochy modelu z tenké uhlíkové tyčky. Tento problém by zřejmě bylo možné vyřešit, ale už i dosud vynaložené úsilí je nemalé.  
3. Kvůli budoucí změně konstrukce ocasní plochy by bylo výhodné, kdyby existovala středová tyčka pod pylonem na kterou by bylo možné uchytit otočnou směrovku umístěnou pod rotorem. Tato úvaha vede k redukci počtu stojin v pylonu ze čtyř na tři. 

Kvůli výše zmíněným bodům bych proto navrhoval konstrukci pylonu změnit na jednotrubkovou verzi. :) Vím že podobné řešení měl i původní TF-G1. V jehož případě vznikal problém zejména s použitím prerotátoru a s pevností spoje mezi vertikální stojinou a podélnou osou trupu. 
Domnívám se, že u malé verze TF-G2 by tyto potíže vznikat nemusely. Neboť nemá prerotátor a spoj mezi vertikálním pylonem a trupem je stejně realizován silentbloky, což je řešení, které by myslím bylo vhodné zachovat. 

Potřebná změna by tak byla pouze v úpravě spodního montážního rámečku pro pylon tak, aby obsahoval kříž do jehož středu by bylo možné vertikální pylon uchytit. Nahoře u rotorové hlavy by trubka mohla být vetknuta do nejtlustšího místa TFSLOT. 

Toto nové řešení by mohlo mít výhodu i ve snazším utěsnění pláště vírníku proti vodě. Neboť na jednoduché vertikální trubce, může být posuvný díl utěsněný o-kroužkem, kterým se přikryje otvor v textilním plášti vírníku. 

Další potenciální výhodou by pak mohla být možnost vedení kabeláže uvnitř trubky. To by mohlo být významné zejména v situaci, kdy by se použila trubka o průměru 13mm, která by měla být dostatečná pro prostrčení prodlužováku servokabelu.

kaklik, 2021-07-06:
K vedení kabeláže z pylonu existuje ještě alternativní možnost na trubku vytvořit nacvakávací 3D tištěný kryt s tvarem NACA profilu, kterým budou kabely svedeny do prostoru trupu.

kaklik, 2021-07-15:
@roman-dvorak  udělal úpravu konstrukce pylonu rotorové hlavy, která pravděpodobně všechny zmíněné potíže řeší. 

[image]

[image]

[image]

Hlavním zádrhelem tak je nedokončená realizace vyměnitelného dílu lámací rotorové hlavy #69

kaklik, 2021-07-30:
S novou konstrukcí pylonu bylo již provedeno několik letů bez zjevných nedostatků.

closed 2021-07-30 by kaklik