ThunderFly-aerospace/TF-G2 #84: Vypadávaní servokonektorů z autopilota

kaklik, 2020-12-03:
U autopilota CUAV V5 nano je poměrně snadné, aby došlo k vypadnutí servokonektorů. Na rozdíl od CUAV V5+ zafixování konektorů nelze jednoduše řešit použitím stahovacího pásku přes přívodní kabely. 

Řešením možná je vytvoření 3D tištěného krytu, který by bránil nechtěnému vypadnutí konektorů. Zároveň by mohl zajišťovat, že konektory budou vždy zasunuty do správných pozic.

roman-dvorak, 2020-12-03:
Nechápu, proč to nejde udělat jako u CUAV V5+? Myslím, že to lze udělat úplně stejně velmi snadno. 

Rozdíl, mezi CUAV V5+ a Nano v tomto je pouze směr hřebínku. Konektor je stejný.

kaklik, 2020-12-25:
ok, můžeme to zkusit. Určitě to bude rychlejší, než kreslit model fixačního dílu. 

Rozdíl já vnímám v tom, že vertikálně vyvedený hřebínek nad sebou nemá plochu ke které by se všechny vodiče daly přitáhnout.

roman-dvorak, 2020-12-25:
Teď předpokládám, že všechny dráty vedoucí do pylonu budou přitažené ke
spodnímu rámečku pylonu. V ideálním případě bude nad tímto místem spojka
konektorů.

roman-dvorak, 2021-01-28:
Prozatím vyřešeno přeorganizováním drátů; další úpravy budou v rámci úpravy pylonu.

closed 2021-01-28 by roman-dvorak