ThunderFly-aerospace/TF-G2 #85: Hrazda usnadňující roztočení rotoru provázkem

kaklik, 2021-01-22:
Z posledních pokusů vyplývá, že by bylo vhodné, aby pylon, nebo rotorová hlava ve svojí vrchní části obsahovala "hrazdu" přes kterou se přetáhne provázek namotaný na bubínek rotoru. 
Důsledkem by mělo být, že k roztočení rotoru by stačil tah za provázek především směrem dolů. Nebyl by tak nutný druhý člověk, který nyní tahá za provázek a drží rotorový pylon, aby jej nevyvrátil do boku. 

Hrazda by mohla být tvořena i bezzávitovou hladkou částí šnoubu. Ve složitější implementaci by byla lepší kladka s kuličkovým ložiskem. Výhodou použití ložiska s drážkovanou kladkou by bylo, že vydrží i mírný nesouosý tah bez snížení účinnosti.

_Originally posted by @kaklik in https://github.com/ThunderFly-aerospace/TF-G2/issues/58#issuecomment-753371874_

kaklik, 2022-05-17:
Vypadá to, že to nakonec vyřešil @slimonslimon  úpravou dílu rotorové hlavy. 

[image]

Myslím že to bude fungovat, ale uvidí se jak dlouho vydrží vytištěný plastový díl tření od provázku. Možná zde bude potřeba přidat nějaký kovový díl, který se provázkem nepropálí.

roman-dvorak, 2022-05-19:
Z pokusů se ukázalo, že systém docela funguje a vítek to dokázal sám roztočit na otáčky, kdy rotor začal mít "zvuk". Zřejmě je to tedy funkční. 

Opravdu problém je, že provázek dokáže plast ubrousit a při třech pokusech (jednou provázek, 2x stuha) se v plastu vybrousila drážka a boční zarážka skoro odpadla. 

Vítek má v plánu tam vmodelovat otvor na šroub, který bude vodorovně umístěn. 

Trochu se obávám, že to nebude úplně dostatčné a a že se plastové Učko bude vybrušovat do stran.

slimonslimon, 2022-05-20:
už mám nakreslenou úpravu kde je šroubek jako osička a na něm vytištěná kladka. ovšem kladka se mi nepodařila vytisknout. (výtisk nebyl symetrický, a nevešel se do prostoru pro kladku) zřejmně ji budu muset ještě rozdělit na půl. 

[image]

dokonce by tato tištěná kladka mohla být navlečena na kluzném pouzdře používaném ve free flapu.

kaklik, 2022-10-15:
Tohle mi přijde jako už vyřešené. :)

closed 2022-10-15 by kaklik