ThunderFly-aerospace/TF-G2 #86: Uchycení podvozkových nohou

kaklik, 2021-01-25:
[Aktuální sestava](https://github.com/ThunderFly-aerospace/TF-G2/blob/4s/CAD/src/assembly/full_assembly.scad) v místě, kde mají být uchycené podvozkové nohy vypadá takto: 

[image]

Protože podvozková noha je část, která se bude pravděpodobně i v budoucnu poměrně často měnit, tak je zřejmě vhodné její uchycení konstruovat tak, aby byla snadno vyměnitelná. 

Z toho mi vyplývá že by neměla být uchycena na pouzdro akumulátorů jak bylo původně v plánu, ale na rohový spoj bočnice a spodní podložky.  Protože bočnice i podložka v tomto místě snadno praskají, tak je nutné oba díly vyztužit.  

Toho lze dosáhnout konstrukcí nového dílu, který bude přišroubován k bočnicím, tak že vyztuží spoj bočnice a podložky.  K tomuto dílu bude teprve šroubována podvozková noha. 

Benefity tohoto řešení jsou: 
-  Není nutné, aby podvozková noha měla rastr děr přesně shodný s rastrem bočnic
-  Fakt že montážní adaptér podvozkové nohy bude šroubován k bočnici lze využít k zajištění matice pro podvozkovou nohu proti bočnici tak, že není možné, aby po odšroubování podvozkové nohy matice vypadla
-  Adaptér může na straně podvozkové nohy končit rovnou plochu, není tak nutné, aby podvozková noha měla levou nebo pravou variantu 
- Tvarem adaptéru lze jednoznačně definovat umístění podvozkové nohy na bočnici a nemůže tak dojít k montážnímu omylu.

kaklik, 2021-02-17:
Uchycení podvozkových nohou bude zřejmě potřeba nějak upravit i v místě připojení k trupu. Neboť je to aktuálně nejcitlivější část podvozkové nohy, která je náchylná na poškození. 

[image]

[image]

[image]

kaklik, 2021-02-22:
K ukrucujícím se podvozkovým nohám mám následující návrh jak to vyřešit: 

Stačí ty odlehčovací trojuhelníky  na začátku udělat menší a začít je zmenšovat až třeba v 1/3 délky té nohy. Tím vznikne u kořene té nohy délka, kde se bude postupně ztenčovat ten materiál, protože noha se bude zužovat ale trojúhelníky budou nejdřív chvíli stejný (menší než jsou teď).

Respektive stačí, aby zmenšování trojúhelníků mělo trochu jiný spád, než je zužování nohy. Tím vznikne u kořene nohy více materiálu, který bude schopen přenášet zvýšené zatížení vznikající na páce nohy. 

Aktuální řešení totiž obsahuje problém, že zatížení podvozkové nohy směrem ke kořenu roste, ale neroste průřez materiálu, přes který se síla může přenášet.

kaklik, 2021-05-03:
Aktuální úprava zřejmě řeší potíž s ukrucováním nohy popsané v komentáři výše. 

[image]

Zbývá tak návrh nového dílu, který umožní napojení na trup.

kaklik, 2021-05-11:
> Ještě by mi přišlo lepší pro tisk, kdyby ta základna trojúhelníků byla v rovině se zůžením nohy.

Ano, souhlasím, na to je separátní issue, které zatím nikdo neřeší https://github.com/ThunderFly-aerospace/TF-G2/issues/98

> 
> Ještě mě napadlo, co zkusit nějaký takovýhle nástroj na analýzu problému lámání nohou: https://youtu.be/7pKvMlo5aMw
> Já si totiž myslím, že by tam uvnitř měla být přenosová výstuha.

Nemyslím si, že je nyní přímo problém v té konstrukci podvozkové nohy. Ale takovýhle nástroj by mohl být užitečný i pro jiné díly. Problémem u 3D tisku je ale to, že ten tištěný model není homogenní, takže to nelze počítat s izotropními tenzory.  Možná to ale jde nějak vyřešit, nezkoušeli jsme to. 

V případě podvozkových nohou je ale nyní myslím problém hlavně v tom vůbec mít nakreslené všechny díly. Druhý problém je pak jejich správné dimenzování, ale to už lze udělat metodou pokus-omyl.

kaklik, 2021-05-17:
@roman-dvorak udělal provizorní úpravu pro montáž podvozkové nohy, která ale moc nezohledňuje zdejší původní plán.

[image]

Nicméně zřejmě umožní dočasnou montáž aktuálně navržených podvozkových nohou.

kaklik, 2021-05-18:
Řešení, které bych navrhoval si představuji nějak takhle: 

[image]

Z boku by vypadalo přibližně takto: 

[image]

Jde o to, že šrouby nohy nebudou pokračovat skrz bočnici, ale budou posunuté oproti původnímu rastru na bočnici.  Všechny šrouby přitom mohou být klidně utopené, neboť adaptér (888_2004) může být dostatečně tlustý na to aby to bylo možné.

kittlerm, 2021-05-18:
Problem trochu je, podstavec pro nohu přesahuje tu rozteč bočnice, takže buď jsou šrouby pro nohy ve vzduchu nahoře nebo dole:
[image]
[image]

kittlerm, 2021-05-19:
Tak to mam asi hotove (ackoli je to cunada, kopiruje to celou 1003), akorat vymyslet umisteni:
[image]
[image]

roman-dvorak, 2021-05-19:
Aby to šlo takto použít, tak tam bude potřeba přidat nějaké tolerance okolo té drážky na  bočnice.

kittlerm, 2021-05-19:
> Aby to šlo takto použít, tak tam bude potřeba přidat nějaké tolerance okolo té drážky na bočnice.

jj taky jsem si rikal, i na ty ostatni diry na kryt, dnes to zkusmo vytisknu a kouknu na to.

kittlerm, 2021-05-19:
Mám ale větší problém, že se neobjevil nový gcode v gcodes repu https://github.com/ThunderFly-aerospace/gcodes/blob/TF-G2/refs/heads/mini/edee/gcode/TF-G2_printplate_888_2004.gcode

kittlerm, 2021-05-19:
> Aby to šlo takto použít, tak tam bude potřeba přidat nějaké tolerance okolo té drážky na bočnice.

Nakonec ten okraj zvětšovat ani není třeba:
[image]

Ale už to mam připravené, jen musím ještě doladit výšku, protože M3x12 je málo a M3x14 je moc a nevejde se, následně dochází k odtažení od bočnice, protože šroub nemá kam pokračovat, proto bych navrhoval, aby ty díry seděli na těch dírách s bočnicemi, ale byli použity jen některé. Co @kaklik  na to?

kittlerm, 2021-05-19:
Ano, už to mám odzkoušené a připravené, včetně možnosti nechat matku více "zapustit". Ale montáž není moc příjemná, chtělo by to přepracovat tu úchytku nohy

kittlerm, 2021-05-20:
Podobně jako je u ostatních dílů chybí zaslepení převisů nad šrouby ( např v https://github.com/ThunderFly-aerospace/Openscad_stdlib/commit/dabd1c7fbdc589b41f5c05d1f52508fff8bbd241#r51066808 )

kittlerm, 2021-05-20:
> > Ano, už to mám odzkoušené a připravené, včetně možnosti nechat matku více "zapustit". Ale montáž není moc příjemná, chtělo by to přepracovat tu úchytku nohy
> 
> Nestačilo by na těch úchytech nohy jen dát ty díry na šrouby dál od sebe? Aby tam bylo více místa na šroubování..
> Jiný nápad moc nemám, protože ty podvozkové nohy nejde snadno rozšířit, aby šrouby šly dát více doboku.

Jo říkal jsem si, že by pomohlo rozšířit rozteč děr

kittlerm, 2021-05-27:
@roman-dvorak @kakl : můžete mi prosím ukázat, jak správně přichytit kolečko? Objímka, ložisko nebo prostě na hulváta šroub?

roman-dvorak, 2021-05-27:
No, vzhledem k tomu, že se to kolečko reálně moc neotáčí, tak nebyl problém ho připvňovat šrubem se samojistnou matkou. A podložku z obou stran kola. Zkoušeli jsme to používat s kluzným pouzdrem, ale nebylo to skoro o nic lepší.

kaklik, 2021-05-31:
@roman-dvorak  Mám za to, že @kittlerm  by měl i ve větvi 4s nahradit tvoje [řešení montáže podvozkové nohy](https://github.com/ThunderFly-aerospace/TF-G2/commit/6e39729c614d194ab27217785f6a86a9e54323e9) s [přidanou kostkou](https://github.com/ThunderFly-aerospace/TF-G2/blob/4s/CAD/stl/888_1003.stl),  [novým dílem pro uchycení nohou](https://github.com/ThunderFly-aerospace/TF-G2/blob/8c0f04c6c630bffd450ea20d603912936f5b1760/CAD/stl/888_2004.stl). 
Souhlasíš s tím?

roman-dvorak, 2021-05-31:
Aktuální implementace je taká strašně monstrózní ;) ... ale dobře. Když to pomůže k vyřešení #109 tak rozhodně.

kaklik, 2021-05-31:
> Aktuální implementace je taká strašně monstrózní ;) ... ale dobře. Když to pomůže k vyřešení #109 tak rozhodně.

Podle mě to s vyřešením #109 moc nesouvisí, protože je to úprava, která je v nějaké formě potřeba pro všechny varianty vírníku. 
Ale s tím, že je to řešení od @kittlerm  zatím takové neuhlazené souhlasím. To se ale snáz odstraní ve chvíli, kdy to bude používat víc lidí a přijde se na nějaký lepší tvar.

kittlerm, 2021-06-14:
Můžu zkrátit tuhle část podvozku?
[image]

Přemýšlím, jak ty nohy uchytit do té středové části, a asi bych tam udělal díru jako na šrou, ale zkusil bych to přidělat jen tavným lepidlem:
[image]

kaklik, 2021-06-16:
> Můžu zkrátit tuhle část podvozku?
>

Já moc nerozumím, co to má přinést. Protože i z fotografie mi to přijde, že když ty podvozkové nohy doléhají na sebe, tak jsou akorát o trochu širší než je trup toho vírníku, což je podle mě správně.
 
> Přemýšlím, jak ty nohy uchytit do té středové části, a asi bych tam udělal díru jako na šroub, ale zkusil bych to přidělat jen tavným lepidlem:
> 

Můj odhad je, že by zde mělo stačit jen tištěné poudzro do kterého se zasunou z boku.

kittlerm, 2021-06-16:
Ty nohy pak kloužou ze strany na stranu, nedrží se na středu. Vyřešit se to dá také lepidlem nebo širokým pouzdrem.

kaklik, 2021-06-16:
> Ty nohy pak kloužou ze strany na stranu, nedrží se na středu. Vyřešit se to dá také lepidlem nebo širokým pouzdrem.

V tom případě mi přijde jako dobré řešení zalepení nohou tavným lepidlem do širokého pouzdra uprostřed.

kaklik, 2021-07-20:
Tak k původnímu uchycení podvozku, které je popsané v úvodu issue mám několik nových poznatků.  Aktuální realizace vypadá takto: 

[image]

[image]

Potíž této konstrukce je, že bočnice na kterých drží podvozkové nohy se při zatížení rozjíždí do stran.  @slimonslimon  i @roman-dvorak  tuto záležitost řeší tak, že obě redukce na kterých podvozkové nohy drží propojili závitovou tyčí. 

Podle mě by však bylo lepším řešením  do přední části navrhnout nový díl, který zastane funkci závitové tyče (podrží vzdálenost mezi montážními díly). Další funkce by byla jako aerodynamický kryt a také ochranný kryt konektoru akumulátoru a akumulátoru před nárazem.

kaklik, 2021-07-28:
Zde je ještě fotka aktuálně používané montáže.

[image]

kaklik, 2021-08-02:
Tak tady je ještě doplnění, jak současný způsob uchycení vypadá po havárii

[image]

Při dopadu navíc došlo k vytržení vodičů z napájecího konektoru na straně akumulátoru (na fotografii chybí). Tato situace ale zkomplikovala dohledávání, neboť vírník tak byl po havárii bez vysílače telemetrie.

kittlerm, 2021-08-04:
Co třeba takhle?
[image]

Přichytávalo by se to těmi spodními šrouby do nohou.

roman-dvorak, 2021-08-04:
Připadá mi (pouze z obrázku), že to zablokovalo přístup k napajecimu konektoru. A nevím, jestli to nekoliduje s bateriovým pouzdrem na nějaké větší baterky.

kaklik, 2021-08-04:
Jo, tvarově je to správně. :)

S tím přichytáváním na spodní šrouby od nohou..  Nevim moc jak to myslíš, ale to jsou zrovna šrouby, který se šroubují pěkně blbě, protože se k nim přes tu nohu moc nemůže. 
Tenhle díl by to naopak chtělo mít možnost sundavat nějak relativně snadno. 

Jako nejlepší řešení by mi přišlo, kdyby to šlo na vírník zepředu nasunout po zapojení konektoru. Bez použití šroubováku. 
Myslím že by to šlo udělat tak, že to buď bude mít vhodný tvar s drážkami (které by tak asi musely být i na dílu pro uchycení nohou).  A nebo to tam můžem dočasně přichytávat lepicí páskou (to ale bude mít nevýhodu, že ten díl nenahradí funkci té závitové tyče, která to nyní stahuje.)

kittlerm, 2021-08-04:
> Připadá mi (pouze z obrázku), že to zablokovalo přístup k napajecimu konektoru. A nevím, jestli to nekoliduje s bateriovým pouzdrem na nějaké větší baterky.

Jo to se musí ošetřit, nevím jestli tam dávat parametr výšky z foto dílu na baterky, aby to lícovalo.

> Jo, tvarově je to správně. :)
> 
> S tím přichytáváním na spodní šrouby od nohou.. Nevim moc jak to myslíš, ale to jsou zrovna šrouby, který se šroubují pěkně blbě, protože se k nim přes tu nohu moc nemůže.
> Tenhle díl by to naopak chtělo mít možnost sundavat nějak relativně snadno. Protože je potřeba připojovat a odpojovat akumulátor vytažením z napájecího konektoru. Jak poznamenal Roman výše.
> 
> Jako nejlepší řešení by mi přišlo, kdyby to šlo na vírník zepředu nasunout po zapojení konektoru. Bez použití šroubováku.
> Myslím že by to šlo udělat tak, že to buď bude mít vhodný tvar s drážkami (které by tak asi musely být i na dílu pro uchycení nohou). A nebo to tam můžem dočasně přichytávat lepicí páskou (to ale bude mít nevýhodu, že ten díl nenahradí funkci té závitové tyče, která to nyní stahuje.)

Ok, zkusím se nad tím zamyslet.

kaklik, 2021-08-30:
> Tohle by mělo sedět s baterkou a zasouvat se na spodek 2004 (tj. pomocného držáku s podvozkem), ještě udělám nějakou nacvakávací zarážku:
> [image]

Tohle vypadá dobře!  Myslím že je to celkem řešení problému. Ještě jen musíme asi rozmyslet, jak to tam bude držet.   Paralelní s tím totiž vzniká [tenhle ušitý kryt](https://github.com/ThunderFly-aerospace/TF-G2/issues/39#issuecomment-908228346).  V případě jeho použití by bylo uchycení krycího dílu snadné. Protože na místě by pak držel po zapnutí zipu na textilním krytu. 
Nepříjemné ale je, že pravděpodobně budeme vírník chtít používat i bez takového krytu.

kittlerm, 2021-09-02:
[image]
Je tam ještě nějaká změna vůči původnímu profilu na větvi mini, takže to musím upravit a otestovat variantu bez kupovaných nohou, tam je úplně jiná rozteč šroubů.

kaklik, 2021-09-18:
Při posledním testu se ukázalo, že je nepříjemně komplikované vymontovat díl motoru. Protože vyšroubování šroubů, které drží díl motoru brání roh podvozkových nohou

[image]
.

kittlerm, 2021-10-06:
Musím to ještě upravit, je tam malá vůle na lehký posun a hlavně při použití té varianty bez koupeného podvozku je tam málo materiálu, takže to má tendenci vyskočit. Zkusím zapracovat druhou pojížděcí lištu na druhé straně.

kittlerm, 2022-09-22:
Malinko jsem oříznul rohy - řešení je +/- pišvorc, asi by bylo lepší vytáhnout pozice šroubů z bočnice, ale prozatím takhle:
[image]

kaklik, 2022-09-22:
> Malinko jsem oříznul rohy - řešení je +/- pišvorc, asi by bylo lepší vytáhnout pozice šroubů z bočnice, ale prozatím takhle: [image]

Mě tohle přijde úplně dobrý! :)  Jen to dej všechno do repozitáře tak, aby to příště šlo vytisknout celý a už to nech být. Víc času tomu není třeba věnovat.

kittlerm, 2022-09-22:
Dávno už to tam je, větev issue_86. Pro tisk jsem si vytáhl amf z artifaktů a oříznul v PrusaSliceru, abych to nemusel tisknout celé, ale jinak je to stejné jako z repositáře. Možná jen to dolní oříznutí není potřeba, to by šlo lehce změnit. Nebo udělat nový díl a mergnout 2002 a 2004? Co na to @roman-dvorak ?

slimonslimon, 2022-09-22:
2002 a 2004 bych nemegoval - zvětšilo by to samotnou nohu, a ta se musí dost často měnit. díky tomu přechodovému díu je ta výměna poměrně rychla.

já myslím, že ty úpravy můžeš rovnou namergovat do 4s

kaklik, 2022-09-23:
ok, považuji za vyřešené.

closed 2022-09-23 by kaklik