ThunderFly-aerospace/TF-G2 #88: Otestovat použití TFRPM a TFSLOT na jedné sběrnici.

roman-dvorak, 2021-02-23:
A jak to teď vlastně je s použitím TFRPM driveru na stejné sběrnici jako TFSLOT?  Funguje to, nebo ne? 

Protože k TFSLOT se stejně I2C na pylon bude tahat, takže nevidím moc rozdíl v tom tam vést dvakrát I2C, nebo I2C a servokablík. 

Z hlediska HW by bylo nejlepší řešení jen jedno I2C na které by byly připojeny oba senzory, díky tomu, že TFRPM je průchozí.  Vím ale že dříve jsme s tímto řešením měli potíže. Je tak otázka zda je to však relevantní i pro současný software?

_Originally posted by @kaklik in https://github.com/ThunderFly-aerospace/TF-G2/issues/76#issuecomment-780865480_

roman-dvorak, 2021-02-23:
Tady asi chybí nějaké "měření/pokusy" který by zjisti, kde je problém.

Driver neprovádí novou inicializaci RPM senzoru. Ale je potřeba vyzkoušet že reset rpm driveru stačí na obnovení funkčnosti.

_Originally posted by @slimonslimon  in https://github.com/ThunderFly-aerospace/TF-G2/issues/76#issuecomment-781265065_

kaklik, 2021-02-23:
Zatím jsem nerozběhl obě čidla, protože je potřeba změnit firmware, nicméně jsem měřil jak vypadá komunikace po I2C s TFRPM a objevil jsem tam tohle:

[image]

Je nepříjemné, že tam jsou ty krátké zákmity podstatně kratší než běžné bity. Může to způsobovat přeslechy a tím pádem hazardní stavy na sběrnici.

kaklik, 2021-02-24:
Tak jsem asi zprovoznil oba drivery. Nicméně zatím nemůžu potvrdit, zda to funguje, nebo nikoliv. Podezřelé ale je, že TFRPM senzor je nyní schopen dávat nesmyslné hodnoty.

[image]

kaklik, 2021-02-25:
Tak pokud se zapojí TFSLOT do série s TFRPM tak se s docela velkou pravděpodobností na SDA stane tohle: 

[image]

Vznikne to následující sekvencí příkazů: 

    icm20948i2c start -X -b 4

V tomto okamžiku je sběrnice ještě v pořádku. Rozbije se až následujícím příkazem: 

    sdp3x_airspeed start  -X

Pokud se TFASPDIMU odpojí a připojí zpátky, tak se sběrnice zotaví a jsou na ní normální data. 

[image]

kaklik, 2021-03-09:
Po vložení sériového rezistoru 82 Ohm mezi konektor a napěťový translátor na signál SDA se situace stabilizovala a zařízení začalo být použitelné.  Dalšího zlepšení lze zřejmě dosáhnout optimalizací hodnoty rezistoru. Viz issue: https://github.com/ThunderFly-aerospace/TFASPDIMU01/issues/5

Závěr však je, že obě zařízení na stejné sběrnici technicky mohou být provozovány.

closed 2021-03-09 by kaklik