ThunderFly-aerospace/TF-G2 #89: Úprava dat pro tisk na Průša mini

kaklik, 2021-03-19:
[Jednotlivé tiskové konfigurace](https://github.com/ThunderFly-aerospace/TF-G2/tree/mini/CAD/slicing) bude potřeba upravit, tak aby výsledné g-kódy byly spustitelné na tiskárně Průša mini. 

Jsou dvě cesty, jak to udělat: 

- Buď přepsat konfiguraci v každém jednotlivém tiskovém souboru, což je asi z počátku velmi snadné, ale bude vždy stejně náročné při každé změně
- Nebo udělat společnou konfiguraci a tu pak slučovat s konfiguracemi pro jednotlivé modely, [pomocí skriptu](https://github.com/ThunderFly-aerospace/processor3D/blob/master/src/merge_slic3r_conf.sh).

Ten výše uvedený slučovací skript, je vůstupem od @JanKott a myslíme si, že je už velmi hotový, ale zatím nebyl prakticky použit.

kittlerm, 2021-03-19:
Zatím jsem udělal úpravu všech. Změny pro Mini jsem dal do mimi.ini - issue uzavřu až to zkusím vytisknout.

kittlerm, 2021-03-21:
První vlašťovka: 
[image]

kittlerm, 2021-03-23:
Nosná podložka připravená k lepení:
[image]

kaklik, 2021-03-23:
> Nosná podložka připravená k lepení:

Super! :)   Je tam ale dost díra.. ;)

kittlerm, 2021-03-24:
> > Nosná podložka připravená k lepení:
> 
> Super! :) Je tam ale dost mezera.. ;)

mno taky jsem si říkal. Obecně mi přijde, že ta tiskárna jede děsně rychle. Rozhodně rychleji, než při zpracování gcodu z PrusaSliceru. Je to taky možná tím, že se tam dává trochu víc extruderu na spodní a vrchní vrstvě, myslím, mám to zkusit poladit nebo to prostě obrousím (postprocessing) ?

kaklik, 2021-03-24:
> 
> mno taky jsem si říkal. Obecně mi přijde, že ta tiskárna jede děsně rychle. Rozhodně rychleji, než při zpracování gcodu z PrusaSliceru. Je to taky možná tím, že se tam dává trochu víc extruderu na spodní a vrchní vrstvě, myslím, mám to zkusit poladit nebo to prostě obrousím (postprocessing) ?

Pokud ta tiskárna jezdí výrazně odlišně oproti průšasliceru, tak je to potřeba poladit. Protože to podle mě povede k tomu, že bude tisknout nepřesně.  Určitě je tam nějaké nastavení rychlostí, které teď asi nebude optimální. 

Diskusi ohledně tisku těchto spojovaných dílů bych navrhoval přesunout [sem](#90). Každopádně si ale myslím, že když to obrousíš a slepíš, tak ti pak nebudou pasovat rozteče těch otvorů pro šrouby po celé délce.

kittlerm, 2021-03-24:
Ok, zkusím se na ty parametry podívat.

kittlerm, 2021-03-24:
Udělal jsem úpravy rychlostí podle PrusaSliceru (PrusaSlicer 0.20mm QUALITY @ MINI, Original Prusa MINI & MINI+)

kittlerm, 2021-03-24:
Ještě je otázka, zda nepřevzít i ostatní parametry ze záložky Printer?

kittlerm, 2021-03-25:
Provedl jsem upravy rychlosti, je to lepsi, ale neni to stejne, PrusaSlicer dela pomaleji externi perimeter:
[image]

roman-dvorak, 2021-03-25:
To je asi potřeba vyexportovat konfigurační ini od Prusi a porovnat celou konfiguraci ini diffem od @JanKott.

Protože to nastavení se nebude lišit pouze v parametrech ze záložky printer (rozdělení parametrů v některých částech není úplně nejlogičtější)

kittlerm, 2021-03-25:
> To je asi potřeba vyexportovat konfigurační ini od Prusi a porovnat celou konfiguraci ini diffem od @JanKott.

Diff umím udělat, ale já mam spíš problém, že nevím, co mam převzít? Protože jak jsem vyrozuměl: @kaklik nechce, aby se to Prušou slicovalo. Já dokonce nemám ani problém změnit slicer, už jsem to zkoušel a mělo by to jít.

> Protože to nastavení se nebude lišit pouze v parametrech ze záložky printer (rozdělení parametrů v některých částech není úplně nejlogičtější)

Souhlas, ale ten hlavní problém je, že neexistuje něco jako jako essential.ini, kde by těch nastavení nebylo stovky, ale jen ty opravu důležité změny.

roman-dvorak, 2021-03-25:
Ano, Průšův slicer nechceme používat, protože nelze použít v rámci actions (a jiné automatizace generování kódu) 

Průšův Slicer vychází ze Slic3r. A díky tomu by (snad) stále velká většína parametrů měla být kompatibilních.

Možná by bylo dobré, kdybys mohl takové porovnání nějak připravil třeba do google tabulky abychom se nad tím mohli společně pobavit. Co by mohlo být důležité použít z jejich konfiguarace a co naopak. Případně jak takový essetnial.ini vyrobit. Protože takového stavu chceme dosáhnout. Abychom pro konkrétní tiskárny/modely upravovali jen dílčí konfigurace.

kittlerm, 2021-03-25:
> Ano, Průšův slicer nechceme používat, protože nelze použít v rámci actions (a jiné automatizace generování kódu)
To si myslím, že je chiméra, zkoušel jsem to lokálně, nevidím důvod, proč by to nešlo v actions na githubu. 

> Průšův Slicer vychází ze Slic3r. A díky tomu by (snad) stále velká většína parametrů měla být kompatibilních.
> ...
ano, zde je tabulka: https://docs.google.com/spreadsheets/d/1DVKDlwXKmDDPxDcK8vhCZcuokp0WVd9-VvMJ01STOAw/edit?usp=sharing

roman-dvorak, 2021-03-25:
> To si myslím, že je chiméra, zkoušel jsem to lokálně, nevidím důvod, proč by to nešlo v actions na githubu.

Dlouho jsem ho neviděl (to bych asi měl někdy napravit). Ale když jsem to zkoušel naposled, tak tam byl nějaký problém, že mu nešlo dát konfiguraci jakou má používat. 

Tabulka je super.

kaklik, 2021-03-25:
> > To si myslím, že je chiméra, zkoušel jsem to lokálně, nevidím důvod, proč by to nešlo v actions na githubu.
> 

Hlavní problém o kterém vím, je že díky tomu že průša slicer je oddělený od Slic3ru. Tak nám nedovede poskytnout důležité hodnoty - posuny modelů při slicování a i jejich těžiště.

kittlerm, 2021-03-26:
> posuny modelů při slicování

to jako myslíš při "arrange"? Můžeš mu ale vynutit pozici z cmd příkazu.

> a i jejich těžiště

na to je nějaký externí program ne? Nebo slic3r to píše rovnou do gcodu? (to jsem si nevšiml). Každopádně jde forknout ne?

Každopádně vytištěné gcody na MINI jsou podle mého laického přístupu dobré. Odtržení od podložky použitím prusasliceru podle mě nevyřeší. Stávalo se mi to i toho letadélka edge aj. částí, když mají malou plochu na desce.

kaklik, 2021-03-29:
@roman-dvorak  Ještě při řešení tisku modelu 888_1002  @kittlerm  přišel s návrhem, že by se v printplate modelu zapínal tisk brimu tímhel způsobem: 

    //@set_slicing_config(../../slicing/default.ini)
    //@set_slicing_config(../../slicing/brim.ini)

Kde brim.ini by obsahoval pouze změněné konfigurační hodnoty proti default.ini, které by zapínaly brim.  Processor3D by tak měl oba soubory sloučit, s tím že vyhrávají parametry z posledního. 
Já to chápu jako nový návrh způsobu jak zadávat slučování konfigurací pro tisk.  Je pak otázka, jakými dalšími způsoby se pak slučovací skript má volat?

roman-dvorak, 2021-03-29:
V návrhu processoru3d se uvažovalo o tom, že by se parametry jako brim zapínaly přímo z openscadu. Tedy že by nemusel existovat ini jen kvůli Brimu.

kittlerm, 2021-03-29:
Přijde mi, že počítat to v openscadu je zbytečné harakiri, nehledě na to, že by to mohlo posunout těžiště, ne? (mno asi to nebude tak významná položka)

kaklik, 2021-03-29:
> Přijde mi, že počítat to v openscadu je zbytečné harakiri, nehledě na to, že by to mohlo posunout těžiště, ne? (mno asi to nebude tak významná položka)

Domnívám se, že @roman-dvorak  asi nemyslí modelovat brim v openscadu, ale myslí do parametrů sliceru jen přímo přidat zapnutí brimu...  Ale popravdě nevím, jakým způsobem je to v návrhu processor3Duvažováno.

JanKott, 2021-03-30:
Přímo do OpenScadu se dá vložit i parametr, například takto:
`//@set_slicing_parameter(infill_only_where_needed, 0)`

kittlerm, 2021-03-30:
A bude tam teda více řádků?
```
//@set_slicing_config(../../slicing/default.ini)
//@set_slicing_config(infill_only_where_needed, 0)
```

JanKott, 2021-03-30:
Ano, ale konfigurace jen jedna. Takto:
```
//@set_slicing_config(../../slicing/default.ini)
//@set_slicing_parameter(infill_only_where_needed, 0)
//@set_slicing_parameter(fill_density, 1%)
```

kittlerm, 2021-03-31:
Bohužel to ale nefunguje, nastavení jsem snad udělal správně:
https://github.com/ThunderFly-aerospace/TF-G2/blob/mini/CAD/src/printplate/printplate_888_1002.scad
neodpovídá nastavení v gcodu:
https://raw.githubusercontent.com/ThunderFly-aerospace/gcodes/TF-G2/refs/heads/mini/edee/gcode/TF-G2_printplate_888_1002.gcode?token=ARVNL5LALSSNXPCITASPVQDAMRPCW

```
; brim_width = 0
```

kittlerm, 2021-03-31:
Možná by stálo za to, kdyby processor3D udělal do artefaktů to nastavení s popisky, co (jaký soubor) to nastavení změnil.

roman-dvorak, 2021-05-21:
> Možná by stálo za to, kdyby processor3D udělal do artefaktů to nastavení s popisky, co (jaký soubor) to nastavení změnil.

asi nechápu, co to má obsahovat?

kittlerm, 2021-05-21:
> asi nechápu, co to má obsahovat?

Komentáře, ze kterého souboru pochází ta hodnota.

kaklik, 2021-06-26:
> Provedl jsem upravy rychlosti, je to lepsi, ale neni to stejne, PrusaSlicer dela pomaleji externi perimeter:
> [image]

Tento díl jde nyní na Průša mini+ bez problémů tisknout. Funguje tisk i mnoha dalších dílů.  Tudíž bych řekl, že toto issue je již vyřešeno. :)

closed 2021-06-26 by kaklik