ThunderFly-aerospace/TF-G2 #90: Řešení způsobu tisku dlouhých dílů

kaklik, 2021-03-19:
TF-G2 nyní obsahuje dva typy dílů, které nejdou vytisknout na běžné tiskárně. 

- [Spodní podložka trupu 888_1001](https://github.com/ThunderFly-aerospace/TF-G2/blob/4s/CAD/stl/888_1001.stl)
- [Bočnice trupu](https://github.com/ThunderFly-aerospace/TF-G2/blob/4s/CAD/stl/888_1003.stl)

[image]

Vyřešení tohoto problému dost souvisí s issue #89 Neboť by zřejmě bylo nejlepší, aby díly byly děleny tak, aby se vešly na danou tiskárnu.

kittlerm, 2021-03-24:
Rozdělené díly na PrusaMini mají přehyb na svrchní straně, takže lepení může být trochu problematické:
[image]

Prozatím to zkusím zbrousit (jen ten přesah), @kaklik se obává, že dojde k nepřesnosti na rozteči děr, já si myslím, že se pohybujeme v rozumném přesahu (do milimetru), takže to nebude vadit. Úpravu pomocí parametrů tisku budu řešit v #89

kaklik, 2021-03-24:
Zkoušel jsem se zamyslet nad postupem jak dělit 3D tištěné modely, tak aby šly lepit nebo i sešroubovat. Nějak jsem nevymyslel nic chytřejšího než tohle:

[image]

[Přikládám zdrojový soubor nákresu](https://github.com/ThunderFly-aerospace/TF-G2/files/6196600/3D_printed_parts_connection.zip)

Nevím zda matematicky vůbec lze vymyslet více řešení. Neboť potíže s tím jsou tyhle: 

1. tiskárna má kulatou trysku, což znamená že na všem vytiskne radius. 
2. Plocha dílů musí být co největší, aby co nejméně záleželo na kvalitě spojení (nanesení lepidla, kvalita svaření a podobně)

Potíž nastává v tom, že čím více se vytvoří radiusů na jedné straně, tím ostřejší tvary vznikají na druhé. Z toho mi vychází, že tato záležitost má možná i geometricky velmi limitovaný počet řešení. 

Jako nejrozumnější mi ale připadá díl dělit v openscadu, takovým způsobem, že by v modelu byly body, kde může vést spline a podle tohoto spline pak model rozdělit. Konkrétně na modelu bočnice by to pak zřejmě vypadalo nějak takhle: 

[image]

kittlerm, 2021-03-24:
Po jemným obroušení (papírem 1000) vypadá spoj dobře:
[image]

Každopádně pevnostně to bude slabé, předpokládám, že se tam ale vystřídá ukotvení bočnice, takže by to nemusela být taková tragedie?

kaklik, 2021-03-25:
> Nějaký test pevnosti? Nebo u podložky je to zbytečné?

U TF-G2 jedinné díly, které potřebují pevnostní test před letem jsou rotorové listy. Ale ty stejně asi vyrábět nebudeš.  U ostatních dílů se pozná až po přistání/dopadu, jak na tom pevnostně jsou. :)

kittlerm, 2021-04-07:
@kaklik našel způsob jak u Průshy vytvářejí puzzle pomocí modifikátorů: https://blog.prusaprinters.org/print-your-own-3d-puzzles_29817/

kaklik, 2021-05-18:
Další možnost je na dělené díly přidat "domečky" které se vzájemně sešroubují šrouby. Výhoda takového řešení by byla v tom, že by nebylo nutné měnit třeba celou bočnici, ale jen nějaký segment. 
Taková změna by mohla zjednodušit i servis vírníku. Neboť v momentálním stavu vyměnit bočnici vírniku, znamená vírnik úplně rozebrat a složit.   
Je možné, že šrouby by šly kombinovat s tvarovými zámky pro rozvedení sil do větších ploch, což by znamenalo, že na bočnicích by pak nebyly žádné lepené spoje.

kaklik, 2022-01-28:
Ta výhoda dělení bočnic rozebíratelnými spoji, se celkem potvrzuje na posledních nehodách, kdy je zbytečně nutné měnit celé bočnice a podložku když je poškozena jen její část.

[image]

Společně s vyřešením tohoto problému by zřejmě chtělo přejít v málo rozebíraných spojích na použití samořezných šroubů do plastu. 
Velmi by to zrychlilo výrobu nových kusů vírníků a odlehčilo konstrukci.

kaklik, 2022-08-25:
K tomuto issue už existuje implementace dělené bočnice: 

[image]

Zatím to předpokládá lepení lepidlem, což neumožňuje selektivní výměnu. Na spodní části je proto jiné experimintální dělení, které ale zas spotřebovává poměrně hodně materiálu.

[image]

roman-dvorak, 2022-08-25:
> Zatím to předpokládá lepení lepidlem, což neumožňuje selektivní výměnu. 

Tohle řešení defaultně nepředpokládá lepení lepidlem. Cílem tohoto slepení bylo porovnat chování slepené a neslepené bočnice. A mým závěrem je, že to vůbec není potřeba lepit a myslím, že to může být provozováno tak, jak je to na druhé straně tohoto složeného trupu. 

Puzzle zámek v podložce potřebuje zvětšit tolerance (protože je to těžko rozebratelné) a zřejmě ani není nutný "schod" pro zvětšení styčné plochy.

kaklik, 2023-06-02:
Zjistil jsem že tahle technika se jmenuje [Dovetail joint](https://en.wikipedia.org/wiki/Dovetail_joint) a jsou tak dohledatelné další detaily a některé triky jak to udělat dobře.

kaklik, 2023-08-15:
Myslím, že mě napadl vůbec nejlepší způsob, jak vyřešit to napojování na relativně tenkých dílech, kde není moc materiálu na puzlle a podobné zámky. 

Abych tu nemusel kreslit náčrt, tak si k vysvětlení půjčím tenhle obrázek: 

[image]

Podstatné jsou zde zejména spoje e) a i). Tyto dva spoje lze v našem případě zkombinovat bez použití lepidla.  Řešení by pak vypadalo tak, že spojovací díl by byl zapuštěn v úrovni dvou spojovaných dílů. V případě bočnic by to znamenalo, že v místě spoje bočnice vznikne ještě další díl, který v podstatě vznikne oddělením několika tiskových vrstev ze základního materiálu. (Kvůli tomu, aby to bylo možné vytisknout, tak zřejmě z vnější strany bočnic, která je při tisku směrem k trysce)

Při spojování bočnic by se pak všechny díly daly přes sebe a prošroubovaly šrouby. Výhody jsou: 

- Nepotřebuje to lepidlo.
- Nepřidává to žádný materiál navíc (kromě možná několika šroubů)
- Nevzniká zde žádný "tektonický zlom", kde se soustředí vzájemný pohyb a síly z dílů bočnic při jejich namáhání. 
- K výměně stačí jen vytisknout nové díly a znovu je našroubovat na stejné místo. 

Nevýhoda možná je to, že při poškození je dost pravděpodobné, že to zasáhne hned dva díly najednou.  To by ale asi nemělo být tak kritické pokud se vhodně uspořádají tiskové sestavy.