ThunderFly-aerospace/TF-G2 #93: Konstrukce podvozkového kolečka

kaklik, 2021-03-22:
Teď když jsem měnil kolečko na vírníku, tak jsem si všiml, že v místě, kde se kolečko běžně láme jsou špatně připevněny paprsky kola k obrubě. 

[image]

Důvodem pak je že kolečko typicky praskne tímto způsobem: 

[image]

roman-dvorak, 2021-05-28:
Nějak jsem tohle issue přehlédl.

Chce si to tedy pokusit pohrát s nastavením slic3r a případně udělat konfiguraci čistě pro kolečko.

Tohle strašně propaguje simplyfi3d, že mají vyřešené.

kaklik, 2021-05-28:
> 
> Tohle strašně propaguje simplyfi3d, že mají vyřešené.

A jak to mají vyřešené?  
Já si myslím, že je především potřeba, aby ty paprsky byly tištěné jako dvouperimetrové.

roman-dvorak, 2021-05-28:
[image]

Vypnutí volby "Detect thin wall" způsobilo tuto změnu:
[image]

kaklik, 2021-05-30:
> 
> Vypnutí volby "Detect thin wall" způsobilo tuto změnu:
> [image]

To mi přijde jako správně.. Teď se ale asi musí zkusit, jestli každý z těch zdvojených perimetrů je dostatečně tlustý proto, aby šel vytisknout..

kaklik, 2021-06-23:
Na posledních výtiscích kolečka jsou zmíněné potíže opravené. A při posledním testu se kolečko dokonce dříve vyšroubovalo než zničilo.. 

[image]

closed 2021-06-23 by kaklik