ThunderFly-aerospace/TF-G2 #95: Otečení šroubů v rotorovém klobočku

kaklik, 2021-04-12:
[Roztáčecí a snímací díl rotoru](https://github.com/ThunderFly-aerospace/TF-G2/blob/master/CAD/stl/888_4008.stl) byl historicky odvozen od [jinak konstruovaného dílu](https://github.com/ThunderFly-aerospace/Auto-G2/blob/master/CAD/stl/111_1004.stl).   V něm byla orientace šroubů zvolena tak, aby bylo možné snadno zkontrolovat jejich utažení. 

S novou konstrukcí dílu však tato orientace přestala dávat smysl a naopak komplikuje složení a výrobu rotoru.  Šrouby v rotorovém kloboučku by tak bylo vhodné otočit, tak aby uvnitř kloboučku byly matice šroubů a nikoliv hlavy šroubů.  

Důsledkem této změny bude, že všechny šrouby bude možné utáhnout běžným inbus bitem standardní délky.

kaklik, 2021-04-24:
Zde je alternativním řešením vkládání matic  do kapes z boční strany. To by mohlo zmírnit problém vypadávání matic během šroubování, protože z boku je snadné při montáži otvory zakrýt rukou. 

Vhodná konstrukce kapes je použita například v [tomto dílu](https://github.com/ThunderFly-aerospace/TF-G2/blob/4s/CAD/stl/888_1018.stl).

kaklik, 2021-05-16:
Nyní má rotorový klobouček ve větvi [THUNDERMILL01](https://github.com/ThunderFly-aerospace/TF-G2/tree/THUNDERMILL01) kapsy pro matice šroubů

[image]

roman-dvorak, 2021-05-16:
Je tahle úprava nějak mergerovatelná do rolničky v master a 4S větvích?

roman-dvorak, 2021-06-26:
Minulý týden jsem tuto úpravu zmergerovat do 4s. Obsahovalo to řadu kolizí. Jeden kus byl vytištěný a ještě to není vyzkoušené (ani neletově).

kaklik, 2021-07-03:
Vírník s upraveným kloboučkem letěl. Potíže nebyly zaznamenány.

closed 2021-07-03 by kaklik