ThunderFly-aerospace/TF-G2 #96: Odlehčení šroubů pro vaničku akumulátoru

kaklik, 2021-04-18:
Pro uchycení vaničky s akumulátorem se nyní používají [šrouby s rýhovanou hlavou](https://eshop.killich.cz/eshop/spojovaci-material/srouby-zatky/srouby-metricke-ostatni/din-464-sroub-ryhovana-hlava/). 

Tyto šrouby fungují velmi dobře, ale mají dvě nevýhody: 

1. Jsou poměrně hmotné, neboť obsahují hodně materiálu
2. Nejsou dobře dostupné, protože jde o speciální typ šroubu

Řešením by mohlo být použít naopak běžně dostupný [šroub se šestihrannou hlavou](https://eshop.killich.cz/eshop/spojovaci-material/srouby-zatky/srouby-s-sestihrannou-hlavou/din-933-sroub-sestihranny-celozavit/).  K tomuto šroubu by pak bylo potřeba navrhnout klobouček s rýhami, který bude sloužit k utahování. 
Takový návrh může vycházet ze současného [dílu pro dotahování matice rotoru](https://github.com/ThunderFly-aerospace/TF-G2/blob/4s/CAD/stl/888_4007.stl).  Z něj stačí pouze odstranit otvor pro provázek a celý díl zkrátit.  Možná bude potřebné přidat osazení na straně šroubu, které zmenší průměr válce okolo hlavy šroubu, tak aby mohl být šroub utažen i mezi žebra na bočnicích.

kaklik, 2021-05-17:
@llucka Ahoj,  je prosím zde nějaká změna?  @kittlerm  by se myslím totiž tahle úprava celkem hodila, aby při výměně akumulátorů nemusel používat imbus. A nebo shánět speciální šrouby. 

[image]

llucka, 2021-05-18:
hotovo - je připravený printplate pro 4 díly

kaklik, 2021-05-18:
> hotovo - je připravený printplate pro 4 díly

@llucka  Super! :)   

@kittlerm  máš nějaké šrouby M3 se šestihrannou hlavou, aby jsi to mohl rovnou vyzkoušet? Nebo budeš potřebovat nějaké dodat? Asi by se mi hodilo znát alespoň nějaký odhad délky, která by tam mohla pasovat, což se dá vyzkoušet se šroubem s válcovou hlavou.

kittlerm, 2021-05-18:
M3 se šestihrannou hlavou nemám (teda ne ze skladu) nevím jestli mam hledat nějaký v mém pracovním bordelu, ale obávám se, že tam budu mít jen M4, protože hlava určitě bude mít rozměr nějaké půlky jako 5.5 a taký jsem si jist, že jsem do těď nepotkal ;)

kaklik, 2021-05-19:
> M3 se šestihrannou hlavou nemám (teda ne ze skladu) nevím jestli mam hledat nějaký v mém pracovním bordelu, ale obávám se, že tam budu mít jen M4, protože hlava určitě bude mít rozměr nějaké půlky jako 5.5 a taký jsem si jist, že jsem do teď nepotkal ;)

Dobrá, šrouby seženu, ale potřeboval bych nějaký odhad jak dlouhé.  Myslíš že bys prosím mohl ten nový díl vytisknout a zkusit třeba pomocí těch šroubů s válcovou hlavou odhadnout potřebnou délku šroubu, případně nějaké jiné geometrické úpravy modelu? 

V repozitáři to vypadá teď takhle: 
[image]

kaklik, 2021-05-19:
Počká to. 

Ještě mezitím bych navrhoval ten printplate_battery_screw.scad přejmenovat na printplate_888_6002.scad, s tím že by asi bylo hezký, kdyby se pouzdro na baterky mohlo jmenovat printplate_888_6001.scad  místo současného printplate_888_1018.scad. Ale nevim, jestli to nevyrobí moc konfliktů při přejmenování ve všech větvích. 

Ten samotný díl vevnitř `hlava_na_sroub()` by asi bylo fajn, kdyby se mohl jmenovat v angličtině.   Takovému typu šroubu se říká  [knurled thumb screw](https://eshop.i-fasteners.com/e-shop/m3x8-zinc-5-8-knurled-thumb-screws-thin-type-din-653_11622503008.html), takže bych to pojmenoval nějak jako `knurled_thumb()` třeba.

kittlerm, 2021-05-20:
Trochu jsem si to upravil na válcovou hlavu, aniž bych změnil délkové rozměr šroubu a M3x12 nechytne matku. M3x14 v pohodě:
https://photos.app.goo.gl/6fBfQecp4DZAsD1g6

Pokud bychom stáhli výšku vroubkaté objímky o 1mm, pořád by se dala pohodlně používat a bylo by možné použít M3x12. (ověřil jsem to) ale to chycení je malé, chtělo by to tak 1.5mm odebrat:
https://photos.app.goo.gl/ngZ7d45pKtFCFwMp7

kaklik, 2021-05-21:
> Trochu jsem si to upravil (lokálně u sebe) na válcovou hlavu, aniž bych změnil délkový rozměr šroubu a M3x12 nechytne matku. 

Tu úpravu na válcovou hlavu jsi si dělal zbytečně.. Ona válcová hlava jde dát do šestihranné díry pro matici.. ;)

> Pokud bychom stáhli výšku vroubkaté objímky o 1mm, pořád by se dala pohodlně používat a bylo by možné použít M3x12. (ověřil jsem to) ale to chycení je malé, chtělo by to tak 1.5mm odebrat:
> 

Z těch fotografií mi přijde, že aktuální objímka zbytečně sežere velkou část délky šroubu.  Navrhoval bych proto šroub více zapustit do vroubkované objímky, aby nebyla taková délka spotřebovaná objímkou.  Aktuální výška objímky vyhovuje dobře, aby se dala pohodlně chytit na členité bočnici a otáčet? 
Pokud ano, tak by zapuštění šroubu myslím šlo pohodlně zvětšit minimálně o 2mm (aktuálně vidím, že je hlava šroubu téměř zarovnaná z koncem objímky).

kittlerm, 2021-05-21:
> Ona válcová hlava jde dát do šestihranné díry pro matici.. ;)

mno alespoň nebylo zbytečné vstávat... :D

> Aktuální výška objímky vyhovuje dobře, aby se dala pohodlně chytit na členité bočnici a otáčet?

Ano:
[image]

I zkrácená varianta jde ale dobře a ještě kousek by šel ubrat:
[image]

kittlerm, 2021-05-21:
> (aktuálně vidím, že je hlava šroubu téměř zarovnaná z koncem objímky)

Valcova hlava je trochu vyšší, ale to zapuštění jsem neměnil schválně, abych měl stejnou hodnotu délky šroubu i s šestihranem.

roman-dvorak, 2021-05-29:
Z této debaty netuším, jaký je závěr?
Bylo by dobré sem napsat konkrétní požadavky na úpravy. O kolik co zkrátit; popř. zapustit.

kaklik, 2021-05-30:
> Z této debaty netuším, jaký je závěr?
> Bylo by dobré sem napsat konkrétní požadavky na úpravy. O kolik co zkrátit; popř. zapustit.

Já si ten model ještě nevytiskl..  Ale @kittlerm  jsem předal různé délky šroubů M3 se šestihrannou hlavou a ten má už nějakou verzi vytištěnou. Takže očekávám, že @kittlerm  má teď nejlepší podmínky pro to definovat, co je špatně a co je potřeba změnit.

kittlerm, 2021-05-30:
Měla by se upravit na hloubka zapuštění, je tam velká část na šroubu, která je zbytečná. Každopádně zítra to přetisknu a vyzkouším.

kittlerm, 2021-05-31:
@roman-dvorak bohužel, kod [TF-G2_printplate_battery_screw.gcode](https://github.com/ThunderFly-aerospace/gcodes/blob/ThunderFly-aerospace/TF-G2%2F4s%2Fmini/gcode/TF-G2_printplate_battery_screw.gcode) není aktuální pro Mini.

kaklik, 2021-06-17:
> @roman-dvorak bohužel, kod [TF-G2_printplate_battery_screw.gcode](https://github.com/ThunderFly-aerospace/gcodes/blob/ThunderFly-aerospace/TF-G2%2F4s%2Fmini/gcode/TF-G2_printplate_battery_screw.gcode) není aktuální pro Mini.

Tak příčina byla v tom, že původní prinplate neobsahoval nastavení pro slicer. Nyní jsem díl [přejmenoval aby odpovídal ostatnímu  číselnému značení](https://github.com/ThunderFly-aerospace/TF-G2/blob/4s/CAD/src/printplate/printplate_888_1020.scad).  Zároveň jsem i přidal nastavení pro slicer. Takže když generování doběhne, tak si myslím, že to bude možné rovnou vytisknout a vyzkoušet. 

Do knihovny jsem i přidal nějakou základní parametrizaci. Takže hlavní geometrické rozměry by mělo jít změnit [tady](https://github.com/ThunderFly-aerospace/TF-G2/blob/c4e445a07dbc138db634a895908c79127e00e91e/CAD/src/888_1020.scad#L6).  

Až bude vybraná vhodná délka šestihranného šroubu, který tam pasuje a upraveny geometrické rozměry tak, aby to šlo pohodlně používat, tak se už může tohle issue uzavřít. :)

kittlerm, 2021-06-24:
Použití M3x12 je pohodlné, možná pro chycení krabicky baterie trochu zbytečně dlouhý, ale zase se lépe chytá začátek do matičky.
M3x10 také jde, ale musí se trochu víc přitlačit, než se chytne matky.
M3x14 nejde dotáhnout, narazí na stěnu.

Co se týče ovládání, tak je to už trochu na těsno k tomu boku, ale jde to používat.
Větší zapuštění asi není vhodné pro použití M3x10, protože se musí prstem zatlačit dovnitř, aby se to chytlo.

[image]
z leva M3x14,12,10

kaklik, 2021-06-26:
> Použití M3x12 je pohodlné, možná pro chycení krabicky baterie trochu zbytečně dlouhý, ale zase se lépe chytá začátek do matičky.

Co teda použít M3x12 a o kousek prodloužit tu válcovou část aby ovládací kolečko bylo trošku dál od trupu? :)

kittlerm, 2021-06-28:
Ok, upravím a vyzkouším.

kittlerm, 2021-07-01:
Mělo by to být hotovo ve větvi 4s.

kaklik, 2021-08-04:
Ok potvrzuji, šroub M3x12 je adekvátní náhradou dosud používaného šroubu s vroubkovanou hravou. :)

[image]

closed 2021-08-04 by kaklik