ThunderFly-aerospace/TF-G2 #97: Špatná hloubka zapuštění matice v dílu 888_1002

kaklik, 2021-05-03:
Díl 888_1002 má u montážních otvorů nevhodně zvolenou hloubku zapuštění matice.

[image]

V důsledku toho nelze pro montáž dílu použít dva stejně dlouhé šrouby M3x10 nebo delší (v případě montáže podvozku). 

[image]

Úprava afektuje verze 4s i verze mini, popřípadě i THUNDERMILL01 úpravu je tak nejvhodnější provést ve větvi 4s, která je base pro další dvě.

roman-dvorak, 2021-05-08:
Když jsem si teď prohlížel ten díl, tak jsem si uvědomil, že původně měla být trochu zapuštěná hlava předního šroubu. Tak, aby vyčnívaly hlavy obou šroubů stejně.

kittlerm, 2021-05-09:
Lepsi tohle A]:
[image]
nebo tohle B]:
[image]
[image]

v kodu je totiz translace bez parametru = warning:
`WARNING: Unable to convert translate(undef) parameter to a vec3 or vec2 of numbers, in file 888_1002.scad, line 141`

kittlerm, 2021-05-09:
Hotovo, ale meumim to mergnout do větve mini, je tam hodně konfliktů.

roman-dvorak, 2021-05-09:
Tak to zkus commitnout do nějaké nové větve. A z ní pak vytvoříme pull request a vyřešíme ty konflikty.

kittlerm, 2021-05-09:
uz to je v 4s

kaklik, 2021-05-09:
Tak jsem to mergnul do mini..  Nicméně si nejsem jistý, jestli to současné řešení reflektuje připomínku z komentáře https://github.com/ThunderFly-aerospace/TF-G2/issues/97#issuecomment-835508091. V takovém případě bych totiž očekával, že ten šroub více vpředu bude mít v modelu částečně zapuštěnou hlavu.

[image]

kittlerm, 2021-05-10:
Aha, hlava šroubu... ja zapustil matici. Navíc jsem to zkosil, protože mi to přišlo lepší pro tisk, ale pokud to chcete jinak, tak to předělám.

kittlerm, 2021-05-10:
> Tak, aby vyčnívaly hlavy obou šroubů stejně.

To jde ale zařídit jen tak, že odstraním úplně to přední část nebo tam udělám otvor na hlavu šroubu a bude to jako v u matice mít takovou malou plošku... hloubka zapuštění tedy bude stejná jako tenčí síla stěny v [1003](https://github.com/ThunderFly-aerospace/TF-G2/blob/4s/CAD/stl/888_1003.stl)

[image]

roman-dvorak, 2021-05-10:
To zkosení a zapuštění matky je dobré. To určitě nechat. U těch matek je
otázka, jestli je nedat do nějakých kaps. Nebo ten rastr trošku zvětšit.
Jak je to jen do poloviny matky, tak se to strašně blbě dává dohromady.

A zapuštění hlavy šroubu je ještě další úprava. Ano, tak aby to odpovídalo
dílu 1003.

kittlerm, 2021-05-10:
Hotovo:

[image]

kittlerm, 2021-05-10:
@kaklik prosím ještě o jeden merge.

kaklik, 2021-05-27:
> @kaklik prosím ještě o jeden merge.

Ok,  mergnuté do mini to je. Nyní to čeká na vyzkoušení. Pak se to může zavřít.

kittlerm, 2021-05-31:
Za mě to je dobré, akorát se nedá použít M3x6:
[image]

closed 2021-05-31 by kittlerm