ThunderFly-aerospace/TF-G2 #103: [4s] TFPROBE shield against parasit light (from enviroment or direct sun light)

roman-dvorak, 2021-05-14:
Po posledním testu se ukázalo, že aktuální uspořádání rotorové hlavy je stále velmi citlivé na mezeru mezi kloboučkm a senzorem TFPROBE. Řešením by bylo ve spodním dílu rotorové hlavy (888_1012) udělat kruhový rantl (cca 3 mm vysoký) který by tomuto světlu zabránil. 

[image]

> Je to pouze rychlé řešení tohoto problému. Komletní řešení by mělo zároveň obsahovat úpravu TFPROBE aby IR signál byl minimálně modulovaný.

kaklik, 2021-05-14:
Já myslím že jsem to už vyřešil [tímhle commitem](https://github.com/ThunderFly-aerospace/TF-G2/commit/b1ed02f40eaf48e77a5badfcbb5dcf91b27b6262#diff-027be5cde553524c3e09e1c626813e0c32ad6d9f485f891ab9735c5bd35d6f84) včera.  ;)  

Teď je to potřeba vyzkoušet.

kaklik, 2021-05-16:
Tady je to složené

[image]

[image]

Nedokonalostí zatím je kvalita tisku mostu, který je ze spodní strany, lemu, který blokuje světlo.   Zadrhává se o něj hřebínek, tohle místo se zatím po tisku musí vyhladit pájkou nebo pilníkem.  Bude asi potřeba zvětšit toleranci pro tisk (asi nejlepší řešení). 

Zatím se mi ale nepodařilo najit takové světelné podmínky za kterých by to bylo rozbité. :)

kaklik, 2021-06-23:
Při testovacím letu upravené řešení TFPROBE měřilo dobře.

closed 2021-06-23 by kaklik