ThunderFly-aerospace/TF-G2 #104: Collision of code-wheel cap and pylon parts

roman-dvorak, 2021-05-14:
V příapdě větší výchylky dochází ke kolizi mezi rolničkou rotoru a pevnou boční částí pylonu.

V dílu 888_1010 je potřeba zmenšit část nad šroubem tvořící osu rotoru. Zároveň by bočnice mohly být trošku tlustší, protože se zde přenáší celá hmotnost vírníku. 

Provizorní oprava vypadala takto: 
[image]

- [ ] Vyřešení kolize
- [ ] Tlustší boční stěny

kaklik, 2021-05-14:
Nevím přesně co, myslíš tím "zmenšit část nad šroubem tvořící osu rotoru".  Ale domnívám se, že zásadní příčina vzniku kolize spočívá v tom, že 

1. Prodloužily se páky serv, což celkově zvětšilo rozsahy rotorové hlavy
2. Došlo k rozšíření dílu [888_1011](https://github.com/ThunderFly-aerospace/TF-G2/blob/4s/CAD/stl/888_1011.stl), který tak výrazně snížil rozsah v ose roll. 

Díl [888_1011](https://github.com/ThunderFly-aerospace/TF-G2/blob/4s/CAD/stl/888_1011.stl) je mimochodem v repozitáři zjevně v jiné verzi, než která je na fotografii, což není moc dobře.

Každopádně primární směr ve vyřešení tohoto issue, vidím v navrácení dílu [888_1011](https://github.com/ThunderFly-aerospace/TF-G2/blob/4s/CAD/stl/888_1011.stl) do původní velikosti zúžením rozestupu mezi horními úchyty.  Další krok pak je sehnat a použít kratší páky u serv a nebo táhla u serv přešroubovat na díru pák nejblíž ose serva.

roman-dvorak, 2021-05-14:
V repozitáři je již novější verze.

kaklik, 2021-06-26:
@roman-dvorak  Není tohle už náhodou vyřešené a ověřené posledním testem? :)

closed 2021-06-26 by roman-dvorak