ThunderFly-aerospace/TF-G2 #105: Change of actuators (servos) in rotor-head and in rudder

roman-dvorak, 2021-05-14:
Nově se budou používat serva [BMS-115HV](https://www.blue-bird-model.com/products_detail/55.htm). Primárním důvodm je, že je potřeba posunout servo na rotorové hlavě trochu dopředu a usnadnit montáž. To aktuálně nelze realizovat z důvodu, že by bylo nutné tisknout vodorovný převis. (nyní se to tiskne na podložku) 

Stejné servo by se pak použilo do ocasu. Kde se to také zjednoduší. Připevňení serv by v obou případech bylo pomocí samořezných šroubků. 

- [ ] Úprava rotorové hlavy
- [ ] Úprava ocasu

roman-dvorak, 2021-05-18:
Ve větvi 4s je tato úprava už implementována. Jen otvor na samořezný šroub je zřejmě moc velký.

kaklik, 2021-05-18:
> Ve větvi 4s je tato úprava už implementována. Jen otvor na samořezný šroub je zřejmě moc velký.

A myslíš tím, že jsou implementovány všechny body zmíněné výše?  Protože ani jeden z nich nevidím zaškrtnutý, což je trochu matoucí.

roman-dvorak, 2021-05-18:
Ocas není implementovaný vůbec. A 4S rotorová hlava má zmíněný nedodělek.

kaklik, 2021-05-18:
> Ocas není implementovaný vůbec. A 4S rotorová hlava má zmíněný nedodělek.

Ok, a co znamená ten bod "úprava pylonu pro master verzi." ?

roman-dvorak, 2021-05-18:
asi to znamená se rozhodnout, která serva se pro master verzi budou používat. Pokud tyto s příčnou montáží (BMS-115HV) tak snížit ten sloupek pro přišroubování a změnit směr díry, kterou se šroubují za přírubu serva. 

Tyto pylony už jsou nemergeovatelné.

roman-dvorak, 2021-05-29:
> Ve větvi 4s je tato úprava už implementována. Jen otvor na samořezný šroub je zřejmě moc velký.

Opraveno.

roman-dvorak, 2021-06-26:
Vypadá to, že u ocasních ploch bude provedeno více úprav. Aktuálně úprava ocasu pro připevnění serva nemá moc význam. Na pár dalších kusech lze řešit tavným lepidlem. 

Umístění serva na rotorové hlavě bylo několika lety otestováno a vypadá jako dobré.

closed 2021-06-26 by roman-dvorak

kaklik, 2021-06-26:
Já bych tu úpravu ocasu pro lepší montáž serva docela rád viděl implementovanou.. To tavné lepidlo se mi nezdá jako moc dobré řešení, měli bychom se totiž už blížit k nějakému velmi definovanému způsobu montáže. 
Také bych rád, aby ocasní servo bylo lépe kryto při nárazu ocasu do země, aby to nebyl první díl o který se vírník zasekne.