ThunderFly-aerospace/TF-G2 #109: Removing branch `mini`

roman-dvorak, 2021-05-29:
Po vyřešení #108 není nutné držet větev `mini`. Netuším, jaké v ní jsou úpravy nad rámec větví `master` a `4S`.

Ví to někdo? @kittlerm 
Pokud tam nejsou žádné úpravy nad ostatní větve (master, 4s), tak bych ji navrhoval odstanit. A nechat jen vývojové větve.

kittlerm, 2021-05-30:
Udělal jsem tam pomocnou [část pro uchycení nohou](https://github.com/ThunderFly-aerospace/TF-G2/blob/8c0f04c6c630bffd450ea20d603912936f5b1760/CAD/stl/888_2004.stl) viz #86, ale můžu to nahrát jinam. @kakl do masteru? V 4s to @roman-dvorak předělal na pevně definované umístění viz [stl](https://github.com/ThunderFly-aerospace/TF-G2/blob/4s/CAD/stl/888_1003.stl)

roman-dvorak, 2021-05-30:
Jo, master je na tohle aktuálně asi vhodná větev. 

(Jinak děláš špatné označení ... Myslíš @kaklik )

kaklik, 2021-05-31:
S větví mini jsem to celou dobu myslel tak, že to bude novější verze vírníku než je master, kterou pak nahradí. Vznikla sloučením větví master a 4s. Větev master je však stále trochu živá, protože ji používá @slimonslimon. 
Větev mini vznikla ohledem na tyhle požadavky: 

- [ ] Použití [TFSLOT](https://github.com/ThunderFly-aerospace/TFSLOT01) pro měření airspeed 
- [ ] V té době hotový návrh pylonu (zároveň dobře opravitelný, protože neobsahuje jiné netištěné komponenty, než šrouby) - Tohle je něco co bych stále nezahazoval, protože na tréning je podle mě tištěný pylon stále lepší, kvůli dobré opravitelnosti havárií.
-  [ ] Úprava nastavení pro tisk na tiskárně Průša mini. 

Vypořádat by se s touto větví zřejmě mělo tak, že se z ní odstraní specifické tiskové konfigurace (stejně jako se asi už stalo se 4s) a pak se z této větve odvodí nová verze vírníku nahrazující aktuální větev master. 
V tomto ohledu bych asi navrhoval buď udělat novou větev TF-G2 a do ní sloučit obě varianty pylonů.   A nebo udělat větve TF-G2 a TF-G2+, které se budou lišit použitím uhlíkových dílů (Na pylon a případně podvozek.).  Jinak by ale měly být kompatibilní a TF-G2+ by mělo vzniknout upgradem z TF-G2.

kaklik, 2021-06-12:
Myslíte, že je lepší mít celotištěnou variantu pylonu jako separátní větev, nebo jako jinak pojmenované díly ve stejné větvi?

kaklik, 2021-06-16:
> Udělal jsem tam pomocnou [část pro uchycení nohou](https://github.com/ThunderFly-aerospace/TF-G2/blob/8c0f04c6c630bffd450ea20d603912936f5b1760/CAD/stl/888_2004.stl) viz #86, ale můžu to nahrát [...]

Změnil jsem úpravu ve 4S od @roman-dvorak  a nahradil jí tvým mezikusem, kterému jsem dále zlepšil geometrii. 
Tím bych řekl, že tahle práce je z mini zachráněněná. Zbývá rozhodnout jak zacházet s variantami pylonu.

kittlerm, 2021-06-16:
Jen rozdělení 1008 a test_spike.scad, takže už asi nic.

kaklik, 2021-06-16:
> Jen rozdělení 1008 a test_spike.scad, takže už asi nic.

Ok, to rozdělení 1008 je záležitost související s pylonem, která asi může ještě chvíli počkat. Minimálně do doby, než se nám podaří navázat nějakou spolupráci s [Techneciem](https://technecium.org/)

kaklik, 2021-08-02:
Fajn, tak bych řekl že vzhledem k pokroku v #82, větev mini už má zachráněné všechny důležité změny. 
Nevím ale co se s takovou větví teď dělá, protože smazat jí mi přijde blbý. :)

closed 2021-08-02 by kaklik

kittlerm, 2021-08-02:
Proč blbý? Klidně bych ji smazal.