ThunderFly-aerospace/TF-G2 #114: ThunderMill - Generation of Kicad edge layer from OpenSCAD

roman-dvorak, 2021-07-22:
Aby bylo možné vyrobit desku, která bude vycházet z OpenSCAD návrhu, je potřeba dodržet následující postup. Je potřeba, abyste se @ChroustJan a @Alsa-Tucnak dohodli, kde kterou část udělá. Já bych navrhoval, aby @Alsa-Tucnak  končila vygenerováním SVG stop. 

Kroky: 

- [ ] Namodelování prototypu v OpenSCADU - @Alsa-Tucnak 
- [ ] Otestování vytištěných disků (sestavení) - @Alsa-Tucnak 
- [ ] Export dat pro import do kicadu  - @Alsa-Tucnak 
- [ ] Import do Kicadu - @ChroustJan 
- [ ] Dokončení PCB - @ChroustJan 

Postup: 

Namodelování v OpenSCADu:
[image]

Vytvoření řezu
[image]

Uložit jako svg:
Soubor -> Exportovat -> Exportovat jako SVG

Otevřít Kicad - PCBnew
Vybrat ořezovou vrstvu
Soubor -> Import -> Grafika

[image]
Zde je potřeba vybrat správný soubor, Vybrat vrstvu pro frézavání (edge.Cuts), měřítka 1,0. 

Následně disk umístit do prostoru

V 3D náhledu bychom měli vidět toto:
[image]

Pokud bude potřeba definovat i plochu výplně, tak to se musí vytvořit vlastní openscad modul obsahujíci plochu měděné výplně˙
[image]

[image]

[image]

A tento objekt se chová jako měděná výplň. To se může hodit na přesné namodelování prostorů elektrod.

kaklik, 2021-07-23:
[Zde jsem našel pána na youtube](https://youtu.be/hHURDpH471o?t=194), který má trik na to, jak díry v DXF zarovnat s footprinty děr tak, aby se ten střed nemusel hledat ručně.

kaklik, 2021-08-12:
Issue přesunuto sem: https://github.com/UniversalScientificTechnologies/THUNDERMILL01/issues/5 Aby jej bylo možné zařadit do projektu.

closed 2021-08-12 by kaklik