ThunderFly-aerospace/TF-G2 #123: Rotor failure - Drumroll emergency mode

kaklik, 2021-11-26:
The autogyro main emergency mode is continuous autorotation flight using the main rotor. The problem emerges in the case of main rotor failure (seizure of bearing, shaft/pylon cracks, etc. ). 

TF-G2 construction with the absence of a horizontal stabilizer is capable to do ["drumroll landing"](https://www.youtube.com/watch?v=i669PHLPcq8). It is rotation mode that effectively slows the descent rate by converting a significant portion of potential energy to aerodynamically damped rotation. 
The net result is significantly decreased impact speed compared to crashes where spinning is not present. 

Unfortunately, this mode needs to be enabled by immediate deactivation of the main engine. This should be done by autopilot, because the human pilot is not capable of that without notice of previous signs of rotor failure. 

Therefore the FC should deactivate the main engine if multiple signs of rotor failure are present. 

- Rotor RPM is zero 
- Rotor vibrations are not present 
- Level flight cannot be achieved (Acceleration is close to zero - free fall)

roman-dvorak, 2021-11-26:
Super.. myslím, že bychom měli začít vytvořením nějakého drumroll detektoru.

kaklik, 2021-11-26:
> Super.. myslím, že bychom měli začít vytvořením nějakého drumroll detektoru.

Já doufal, že tohle bude řešit někdo úplně jinej. Proto jsem to psal anglicky. :D

kaklik, 2023-08-16:
Kromě přistání, tak i pro tento účel by se hodilo, kdyby vírník měl za letu odpojitelný rotor.  Jeho odpojením by bylo možné vyřešit následující situace: 

- Okamžité ukončení letu v kritické situaci (např. bezpečnostní situace, kdy hrozí kolize, nebo vstup do zakázaného prostoru)
   - Jde tím otestovat a následně použít relativně bezpečný mód pádu v ploché vývrtce, jak je popsáno výše
- Zjednodušení algoritmu přistání. Neboť odhozením rotoru těsně nad zemí se vyřeší problém "co s energií v rotoru po dosednutí".

Technicky jsou dva možné způsoby realizace. První je použití serva pro odjištění ložiskového sloupku rotoru. Druhá možnost je tepelné oddělení plastové pojistky, která bude držet sloupek rotoru.

V případě použití serva, je softwarová realizace celkem přímočará, protože lze použít přímo některý volný PWM výstup. V případě tepelného aktuátoru řešení už tak jednoznačné není, protože PX4 na to nemá přímo vyřešený ovladatelný výstup. Bylo by nutné použít nějaké náhradní řešení například přes  [camera trigger](https://docs.px4.io/main/en/peripherals/camera.html#trigger-interface-backends). [Padáková logika implementovaná v PX4](https://docs.px4.io/main/en/peripherals/parachute.html) zdá se problém nijak neřeší, prototože předpokládá použití PWM výstupu v režimu serva.