ThunderFly-aerospace/TF-G2 #132: Lightweight motor holder

kaklik, 2023-08-14:
[image]

3D tištěný držák motoru by se zřejmě dal výrazně odlehčit.  Navrhoval bych do něj udělat díry, které by zároveň snížily spotřebu plastu na jeho vytištění. Aktuálně je potřeba držák vyměnit na obou havarovaných vírnících. 
Hmotnost současného držáku je ~40g.

Možná že lze odebrat i celou středovou část, která spojuje díry pro šrouby motoru. Je s tím akorát potíž, že bude obtížné takový tvar vytisknout.

roman-dvorak, 2023-08-14:
Já jsem o tomhle také přemýšlel. A ještě mne napadlo, že by se mohl využít duralový kříž, který je s motorem dodáván. Sice to přidá další tři šrouby ale ve výsledku by to mohlo být ještě lehčí.

kaklik, 2023-08-16:
Současné upravené řešení je dostatečné.

closed 2023-08-16 by kaklik